OrcaSlicer/OrcaSlicer #3043: Orcaslicer crash if use pressure advance test

luco85, 2023-12-08:
### OrcaSlicer Version

1.8.1

### OS version

OSx Sonoma 14.1.2

### Additional system information

Apple Macbook M1 Pro
16gb RAM

### Printer

custom

### How to reproduce

go to calibration, pressure advance, choose Tower PA or PA pattern, press OK
It work only with Line PA

### Actual results

after press OK to generate the app crash.

### Expected results

the PA pattern calibration should appears to te bed

### Project file & Debug log uploads

[debug_Fri_Dec_08_16_39_36_20431.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/13617537/debug_Fri_Dec_08_16_39_36_20431.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

luco85, 2023-12-08:
update: if I have a default line width set to 0, orca crash. if I put a 0.4 value and go to PA test, I select pattern PA and click OK, it ask to transfer new value to profile and all work fine. Orca set his value for the test and all works.

luco85, 2023-12-09:
here! if I launch pa test from this conf it crash (but not if I choose the line method). If the value is not 0 all work and PA test set his value to line widh

[image]

Boerny41, 2023-12-09:
Can confirm this problem. A small rectangle, 4 layers high, is also generated: 
[image]

> Why would you put a default line width to 0?

Why not? I've only set the top line width because I want to force it to 0.4. For the rest, Arachne can figure it out.

igiannakas, 2023-12-09:
This is not the intended use -
Default line width if other line widths are set to 0. If expressed as a % it will be computed over the nozzle diameter.

You need to set this parameter to a value appropriate to your nozzle size. Arachne won’t calculate line widths unless it’s needed for a particular path - it will use the default or specified line widths as the starting point and adjust only if necessary.

if you set the values to 0 as above the slicer will default them internally to 0.45 for all (112.5%).

hence you’re not gaining anything by not specifying them - you’re just setting them to 0.45…. While this is a bug in the PA test sure, it’s assuming that the user has set the default line width to something. You can set it to 0.45 and nothing will change in your settings.

igiannakas, 2023-12-09:
Ive pushed a fix to set the value to a sensible default to prevent the PA test from crashing; however you shouldn't be using 0 on the default extrusion width as this is assumed as the default if no other line width is set and there may be other areas where this is assumed to be non-zero in the slicer.

igiannakas, 2023-12-09:
The rectangle is intentional. It’s the have a representation in the model screen.

luco85, 2023-12-09:
ok thanks!

igiannakas, 2023-12-10:
Fixed in the latest nightly. Please test and close issue.

luco85, 2023-12-10:
Need to re download and reinstall?

luco85, 2023-12-10:
seems it work fine now! thanks

phoenix79, 2023-12-14:
I've been having the same issue. I have not tried the nightly, but I'm assuming that you've set it to use the nozzle diameter if the Default line width is set to 0? Not just hard coded to 0.4?

igiannakas, 2023-12-14:
Have set it to 112.5% of the nozzle diameter which is the default handling of line widths if set to 0.

phoenix79, 2023-12-14:
> Have set it to 112.5% of the nozzle diameter which is the default handling of line widths if set to 0.

Thank you!

phoenix79, 2023-12-14:
Nightly 231210 (linked above) seems to still have this issue for me. Set Default line width to 0 and try a PA Tower test, I get an Unhandled Exception error. Put a value in Default Layer Height and it works fine.

closed 2024-03-21 by github-actions[bot]