OrcaSlicer/OrcaSlicer #3050: Slicer adding filament change gcode at start of print

53Aries, 2023-12-09:
### OrcaSlicer Version

1.8

### OS version

Windows 11 

### Printer

Custom klipper

### How to reproduce

Single extruder multimaterial
No purging or ramming enabled
Line 125 and 126 should not be there
[image]

[image]
[image]

### Actual results

After initial load the filament change gcode is called

### Expected results

No toolchange or filament change gcode should be inserted. I don't know if I'm just doing something wrong or what. 

### Project file & Debug log uploads

[Cube.zip](https://github.com/SoftFever/OrcaSlicer/files/13623351/Cube.zip)
[Cube_PLA_18m0s.zip](https://github.com/SoftFever/OrcaSlicer/files/13623353/Cube_PLA_18m0s.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13623355/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

RayTitone, 2023-12-10:
were you able to solve this? as im having the same issue on my k1. im very new to klipper and not to sure what im doing

53Aries, 2023-12-10:
> were you able to solve this? as im having the same issue on my k1. im very new to klipper and not to sure what im doing

So here's what I'm doing as kinda a work around. 
[image]
[image]
So what happens is on the print start TR will load the initial lane and use my custom purge macro. After thats done it kinda jumps around and wipes following the go to purge and wipe macros but the duplicate T(x) is ignored since it's all ready on that tool. For the next tool change everything works as it should. 
If you are using Tradrack check in the discord for my thread in the build help for more specific info. I just assume you are since you have a K1.

closed 2024-03-18 by github-actions[bot]