OrcaSlicer/OrcaSlicer #3051: Adopt Settings when Nozzle diameter is changed

awiltschek, 2023-12-09:
As it is possible in Cura i would like to see that all relevant parameters according to the nozzle size are changed fully automatically and can be saved according the nozzle diameter.
It would also help when you can select the nozzle diameter as an additional selection criteria like in cura so you can adjust the impacted values automatically (eg Linewith, Layerheight, ....)

At the moment this is a missing major feature to be a full replacement of cura.

I realy love your slicer but transferring now all settings for my sidewinder from Cura to Orca for a 0.6 nozzle or a 0.2 nozzle will not help me in getting rid of cura.

So for this diameters i still would need cura which means i cant switch to orca as i very seldom print with 0.4

igiannakas, 2023-12-09:
> As it is possible in Cura i would like to see that all relevant parameters according to the nozzle size are changed fully automatically and can be saved according the nozzle diameter. It would also [...]

You can specify the line widths as a % of the nozzle diameter. Then all you need to change is the layer height and first layer height.

awiltschek, 2023-12-09:
thnx for the tip will try this until my feature request was analyses and implemented.

if a new slicer wants to gain market share it should first include all features which are needed for normal users (non expert) and then address the needs of print-nerds/pros :-)

inspectionsbybob, 2023-12-18:
I agree, there needs to be something "better" as I created a printer with a 0.6mm nozzle and it took a while to find -ALL- of the 0.40 settings....

awiltschek, 2023-12-19:
the tip with the % usage makes it work for me although 0,66 makes it more visbile for me.
i think splitting it up in ...
printer - settings
nozzle - settings
filament - settings
print - settings

would be the clearest settings-structuring

i realy love that most of the settings are now there where they get influenced.
in cura material is worthless for me.
i have for each of my material an own profile (and jep then i change nozzle size and the rest is done by cura)

but nozzle - settings is somehow what missing in this new slicer

and some better gui handling of all saved variants . at the moment i have too many filaments, settings, printer entries 😂

closed 2024-03-27 by github-actions[bot]