OrcaSlicer/OrcaSlicer #3072: Chamber control Qidi X-Max 3

cmtequites, 2023-12-10:
Chamber temperature control is not working properly with Qidi X-max 3, even after adding M191 as instructions.

Orca define as "chamber heater"  Qidi define as "hot" inside Printer.cfg

after replacing this names the chamber control works partially, when you add it to the top of the g-code file as instructed the printer display Unknown command:M191 messages on the console tab, when you add it after M141 for example the chamber heater starts before bed mesh, when the selected temperature is reached, chamber heater turns off and remains off for the rest of the print.

Phil1988, 2023-12-13:
I read your description 5 times now - and I still dont know what happens when and what...
I hope the developers can understant what you reported here.

Could you maybe make it more clear by eg. structure it better and writing in closed sentences with periods?

cmtequites, 2023-12-13:
Ok Phill, let me try to explain it better. 

Using orca with the X-MAX 3, you need to insert a G-code inside the "printer.cfg" file in order to use the chamber temperature auto control. 

Orca provided this g-code inside the tab "notes".

This G-code is partially wrong. Orca defines chamber control as "chamber heater", the Manufacture call is it "hot". 

Replacing this nomenclature does not solve the problem, once the print starts, chamber start heating to the selected temperature even before bed leveling occurs, that is not the right order. 

After the temperature reaches the selected value, chamber heater turns off and remains off for the rest of the print, unless you re-select the value you want.

Phil1988, 2023-12-14:
This describes it very good 👍 

For the qidi you can not really rename it from "hot" to "chamber_heater", as it will break your screen funcionallity on this regard.
What you basically want is that orca puts out a 
`M141 S55 ` 
After homing etc. and before the print right?

Can you use something like this in your gcode start script?
`M141 S{chamber_temperature} ; set chamber temperature manually

PhavourFlavour, 2024-01-04:
> This describes it very good 👍
>
> For the qidi you can not really rename it from "hot" to "chamber_heater", as it will break your screen funcionallity on this regard. What you basically want is that [...]

I have a Qidi X Plus 3 and trying to make it work properly with Orca, too!

I tried setting `M141 S{chamber_temperature}` manually,

but this leads to  an error when slicing by Orca:

```
machine_start_gcode Parsing error at line 4: Referencing a vector variable when scalar is expected
```

do you know how to convert such value types within gcode?

nicolasavru, 2024-01-13:
I think it wants square brackets instead braces: `M141 S[chamber_temperature]`
The default machine start g-code had a few similar lines that used square brackets, for example `M190 S[first_layer_bed_temperature]`. I inserted the M141 line right before the M190 line and it seems to work fine.

chucky83, 2024-02-17:
use M141 S[volume_temperature] it works on mine

GrubbyZebra, 2024-02-21:
Can confirm this works:

1) Copy and paste the M191 code from OrcaSlicer into the X-Max 3 printer.cfg file (copy and paste, don't try to type it manually as that is prone to error)
2) Edit macro by changing chamber_heater to hot (all lowercase) in _both_ places in the macro
3) Restart Firmware
4) Select "active temperature control" option in OrcaSlicer and enter desired chamber temp in the appropriate field.

This will insert the M191 command into your g-code, and the print will start by heating the chamber (and will wait for the temp to be reached before proceeding, which can take a looong time).

GrubbyZebra, 2024-02-21:
Would be nice to have the option to use M141 with the active temperature control option, or have the chamber heater command insert later in the G-code after the extruder temp is set and active, since this greatly reduces the amount of time required to heat the chamber.

closed 2024-05-29 by github-actions[bot]

cmtequites, 2024-06-04:
> Would be nice to have the option to use M141 with the active temperature control option, or have the chamber heater command insert later in the G-code after the extruder temp is set and active, since this greatly reduces the amount of time required to heat the chamber.

It would be perfect!!!

cmtequites, 2024-06-04:
> > Would be nice to have the option to use M141 with the active temperature control option, or have the chamber heater command insert later in the G-code after the extruder temp is set and active, since this greatly reduces the amount of time required to heat the chamber.
> 
> It would be perfect!!!

If they copy the start procedure from Qidi would solve this issue. Chamber heater control should star after the Bed is hot, the temperature from the bed helps a lot with chamber temperature as well.