OrcaSlicer/OrcaSlicer #3076: Restart Print - without requiring start_print gcode to run

Ltek, 2023-12-11:
**Is your feature request related to a problem? Please describe.**
on occasion I (and I'm sure others) forget to clean the plate or something like lock in the filament... and we need to restart the print. But since the best mesh is already calculated and such, we dont need the printer to do that again. Would be great to have a simple Stop & Restart feature to solve this challenge, and save time.

**Which printers will be beneficial to this feature**
Klipper based printers

**Describe the solution you'd like**
Restart print without need to rehome and run Start_Print gcode

**Describe alternatives you've considered**
done know of any that exist

HakunMatat4, 2023-12-11:
This isn't a slicer problem.
I have a LDO Voron Trident, running klipper, this is a printer feature.

Stop, Resume, Restart is a printer thing, if you use KlipperScreen to control and manage your printer, these options are right there

And how can the printer restart a print without homing???
The toolhead has no idea where it is, you must home unless you don't mind the nozzle crashing into the bed.
Starting a print without calling `PRINT_START` is like wanting to drive a car without petrol.

This request is confusing and not a slicer problem.

Ltek, 2023-12-11:
Slicers can and should have this feature IMO. The printer homed at the start and the toolhead knows where it is when it pauses. Klipper has the data it needs from the initial homing, nothing changed. 

The ask is to restart at Layer 1. So the slicer needs to 'Pause and Restart at Layer 1 

Manually editing gcode works to do this (its manual and not fun). Many people (including myself) have done this on failed prints to restart them at the failed layer. 

Please dont thread crap, its disrespectful to everyone. If you are not trying to help figure it out how to make it possible, saying you'd like the feature, or are a dev explaining how it might work/not... then you are thread crapping.

closed 2024-03-19 by github-actions[bot]