OrcaSlicer/OrcaSlicer #3078: [QUALITY] Allow custom bridging angle/infill like SuperSlicer

HakunMatat4, 2023-12-11:
**Is your feature request related to a problem? Please describe.**
N/A

**Which printers will be beneficial to this feature**
Everybody

**Describe the solution you'd like**
Be able to change the bridging angle, allowing it to look cooler when a face with bridging is exposed like so.

[image]

According to the post, these are the SuperSlicer options to archive that nice effect.

```
Print Settings
  -> Perimeters & Shell
    -> Advanced 
      -> No Perimters on bridge areas - Keep Only bridges

  -> Infil
    -> Advanced
      -> Angle 
          -> Bridging - 90 (set this based on parts orientation to make it look the best)

  -> Width and Flow 
    -> Overlap
        -> Bridge Line Density - 50% Min, 50% Max
    -> Flow
        -> Flow Ratio Bridge -  85% 
```

I do understand that "No Perimters on bridge areas - Keep Only bridges" and "Bridge Infill pattern" might no be an easy task but allowing the bridge angle to be changed should be a fairly "easy" implementation.

[image]

That also should prevent the infamous bridging belly.
This is the current way, both layers on the same direction so a lot of material concentrated within the same space.
The example above, the next layer is crossed distributing its weight across many lines instead of 1-2 lines so it is lighter.

[image]
[image]

**Describe alternatives you've considered**
Cry :)

**Additional context**
Pleeeeeeease!!

Noisyfox, 2023-12-11:
[image]

HakunMatat4, 2023-12-11:
@Noisyfox yeah, I need to book an eye check...
I got the first bridge to do that but it only applies to the first bridge

[image]

The second bridge remains the same.
I am guessing because of "50% Min, 50% Max" which OrcaSlicer seems to be only applying one (???)
Any idea why of these weird idk what on, at the corners??
I have noticed these weird stuff everywhere on prints, especially the ones with low wall counts.
I am opening a bug for that one, no logic explains it.

Thank you

[image]

HakunMatat4, 2023-12-11:
"Detect narrow internal solid infill" solved those weird concentric stuff.
Thanks for that Noisy

Ltek, 2023-12-11:
I'd also like this. I have a similar problem with Top/Bottom Layers (Skin), opened a bug report... https://github.com/SoftFever/OrcaSlicer/issues/1515#issuecomment-1794798789

jpbnto, 2024-02-19:
[image]
 i have similar problem 
 
 orca on the left, superslicer on the right. 
 orca makes concentric moves on the bridge.

Noisyfox, 2024-02-19:
> ![image](https://private-user-images.githubusercontent.com/91659388/305968150-afc367cf-d5fb-4382-aa1e-a139e01266f3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Make sure the "extra perimeter on overhang" option is disabled.

jpbnto, 2024-02-19:
> Make sure the "extra perimeter on overhang" option is disabled.

I disabled the "detect overhang walls" and it made it rectilinear movement.

It is not the extra perimeter on overhang. ( Cannot see the difference enabling or disabling this setting or am i not just looking at thr right place?

confuzled-dev, 2024-07-12:
also interested in bridging angle adjustment

jpa00, 2024-09-26:
The superslicer setting mentioned here, "no perimeters on bridge areas" sounds like it would solve or at least let users work around the persistent issues (see for example [#5724](https://github.com/SoftFever/OrcaSlicer/issues/5724)) of bridge perimeters printing with either
- normal perimeter speeds when "detect overhang walls" is disabled, which usually is either too slow or too fast
- 75-100% overhang speeds when "detect overhang walls" and "slow down for overhangs" are both enabled, which almost certainly is way too slow

closed 2025-01-03 by github-actions[bot]