OrcaSlicer/OrcaSlicer #3079: Weird concentric pattern lines instead of just monotonic aka lines

HakunMatat4, 2023-12-11:
### OrcaSlicer Version

1.8.0 official

### OS version

Linux

### Printer

Voron Trident

### How to reproduce

1. Slice anything
2. Check the layer by layer 

### Actual results

These concentric moves shouldn't be happening, it breaks the smooth run with these small runs and extra retraction.

[image]

These concentric prints will break the top layer, it isn't smooth lines affecting the top layers on top of it, I mean, that section top layer looks very bad.

[image]
[image]

### Expected results

Keep the line pattern 

### Project file & Debug log uploads

[orca.log](https://github.com/SoftFever/OrcaSlicer/files/13631809/orca.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2023-12-11:
Disable this 
[image]

HakunMatat4, 2023-12-11:
oh wow, this has been driving me nuts and I swear I do not remember enabling this.
It started showing out of the blue and I had no idea how to disable it.

Thanks a lot.

closed 2023-12-11 by HakunMatat4