OrcaSlicer/OrcaSlicer #3080: Profiles are not being synchronized between computers

szafranski, 2023-12-11:
### OrcaSlicer Version

1.8.1

### OS version

Mac Sonoma and Windows 11

### Additional system information

Hi, I use the slicer on two separate computers. I have an issue because some printer profiles are not being synchronized between them. I am logged into the cloud, I tried to reinstalling on both computers. 

### Printer

Prusa i3 and Creality 10S

### How to reproduce

Log in

### Actual results

Not all printers are being synced

### Expected results

All profiles should be present on all computers

### Project file & Debug log uploads

[End_stop_z1.zip](https://github.com/SoftFever/OrcaSlicer/files/13633539/End_stop_z1.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13633541/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MustDie-green, 2023-12-20:
Ventura 13.4 and Widows 10 the same problem

themaninthesuitcase, 2023-12-21:
Sonoma 14.1.1 and Windows 10, same issue.

Profile created on the mac and isn't appearing in Windows.

Log from source orca attached, 

Seeing this in my logs and the above could this be the cause?
`[2023-12-14 22:31:53.588691 +00:00] [E] [T 2661482]: bambu_network_connect_server , connect_server returned error:-1`

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13745190/log.zip)

timboll, 2024-03-13:
I'm also having this issue with Sonoma 14.4. Profiles are syncing fine between windows computers on the same account I'm using with Mac. Other Bambu Studio functionality works fine, like sending prints and monitoring cameras. I'll add a log file once I reinstall.

closed 2024-06-20 by github-actions[bot]

cheshira, 2024-12-21:
Hi, all!
Having same issue here and feeling quite mad about it. No matter what, but I can't make my printer profiles to sync between laptop and desktop. Tried to toggle and untoggle auto-sync checkbox, tried to reinstall Orcaslicer, but sync wouldnt work correctly. It actualy did sync some irrelevant priner profile, but not those, that are actually in use here in the lab.

ZombieEater, 2025-05-10:
Any way to put this issue as active? The non sync still happens.