OrcaSlicer/OrcaSlicer #3083: MacOS: font list in OrcaSlicer is way smaller than the font list in MacOS

netweaver1970, 2023-12-11:
### OrcaSlicer Version

1.9.0-Dev

### OS version

MacOS Sonoma, 14.1.2

### Additional system information

Arm M1 - MacBookPro

### Printer

Voron 2.4, single extruder, no AMS

### How to reproduce

install now ttf (or otf) fonts on MacOS level (in my case multiple free TTF Stencil fonts, downloaded from the internet).
They are nicely visible afterwards in FontBook program
But newly added fonts are NOT visible in the OrcaSlicer list of fonts to be used for the text tool.

### Actual results

All Stencil added fonts (Post-no-Bills, Ruler Stencil and Spotlight) are not visible in OrcaSlicer.

All added fonts are validated successfully and usable (eg. Ruler Stencil) in other apps.
[image]

This might be a general MacOS issue, I'm just puzzled why it only is apparant on MacOS. Is there a ways to refresh a program specific cache or something?

### Expected results

All visible fonts in Fontbook should also be visible/usable in OrcaSlicer.

[image]

[image]

### Project file & Debug log uploads

the project file is not relevant, as I can't even select the font needed before saving
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13636090/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mkorczyk, 2023-12-31:
I am having the same issue. I was able to add custom fonts to v1.6.3 but not v1.9.0 beta.  I haven't tried any versions between them though.

NeighNeighNeigh, 2024-03-02:
having the same issue, very annoying

netweaver1970, 2024-06-05:
in the latest official release I can see all fonts, so for the issue is gone. I don't know waht solved it, MacOS updates, some -under water- font DB fix or a newer Orca release. Anyhow, for me it can be closed.

closed 2024-09-12 by github-actions[bot]