OrcaSlicer/OrcaSlicer #3084: Klipper: max_accel_to_decel is being replaced with minimum_cruise_ratio

ericderace, 2023-12-11:
**Is your feature request related to a problem? Please describe.**
A breaking change is coming to Klipper - max_accel_to_decel is being replaced with minimum_cruise_ratio

**Which printers will be beneficial to this feature**
Klipper based printers

**Describe the solution you'd like**
In print profiles, add an option to enable "minimum_cruise_ratio", much like "enable max accel_to_decel".

**Describe alternatives you've considered**
The alternative is to disable max_accel_to_decel in print profiles and set minimum_cruise_ratio in the firmware (printer.cfg) or in additional printer g_code settings (SET_VELOCITY MINIMUM_CRUISE_RATIO=value) 

**Additional context**
[Klipper pull request](https://github.com/Klipper3d/klipper/pull/6418)

dstulken, 2024-03-15:
Not stale. 

The Klipper docs state that the max_accel_to_decel parameter "is deprecated and should no longer be used".

The changes for minimum_cruise_ratio have also now been merged:
https://github.com/Klipper3d/klipper/commit/0105aa330f2a4865b01cbbc548cba1322be5371e

dstulken, 2024-03-15:
I had thought that the stale tag would have been removed after my comment, but it seems this is not the case. Does anyone know how to get this status changed?

vgdh, 2024-03-15:
> I had thought that the stale tag would have been removed after my comment, but it seems this is not the case. Does anyone know how to get this status changed?

maybe @SoftFever knows

SoftFever, 2024-03-17:
Oops, I overlooked this ticket.
Replied here:
https://github.com/SoftFever/OrcaSlicer/issues/4521#issuecomment-2002295479

closed 2024-06-24 by github-actions[bot]