OrcaSlicer/OrcaSlicer #3086: Overhang percentage or angle in the GCode previewer and color scheme

e-dreyer, 2023-12-11:
**Is your feature request related to a problem? Please describe.**
I would like to be able to preview what the slicer calculates as the angle for overhangs. With many designs, I often design with a max overhang angle of 60°. I am however not able to always achieve this and often it is required to decrease the layer height of the slicer to reduce the number of tool paths identified as overhangs (in the dark blue preview). 

Similar to how we can preview fan speed, flow, width etc. I would like to be able to view the overhang angle or percentage of overlap that the slicer calculates. This would help with improving designs as well as provide a better direction to move to when adjusting extrusion width and layer heights to reduce the number of overhangs and improve printing.

**Which printers will be beneficial to this feature**
All printers would benefit from this. It would allow users to see the overlap of the line relative to the previous layers and know whether they should change the angle or dimensions of their design or possibly tweak layer height and extrusion widths.

**Describe the solution you'd like**
The slicer currently allows use to preview Gcode with different color schemes for flow, fan speed, layer time etc. The slicer also allows use to set the dynamic speeds and fan speeds for percentages of overhangs. Thus, it is already available by looking at the print speed, fan speed and flow and comparing this with your settings for different percentages of overhangs.

tome9111991, 2023-12-19:
I'm not sure if it's helpful for you.
But with Support Painting you can Highlight the Overhangs
[image]

But i like your idea +1 for this

e-dreyer, 2023-12-20:
> I'm not sure if it's helpful for you. But with Support Painting you can Highlight the Overhangs ![hoverhangs](https://private-user-images.githubusercontent.com/57866234/291647911-40dbf150-77da-43e3-8042-817cec3dbca5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxMDY2NjksIm5iZiI6MTcwMzEwNjM2OSwicGF0aCI6Ii81Nzg2NjIzNC8yOTE2NDc5MTEtNDBkYmYxNTAtNzdkYS00M2UzLTgwNDItODE3Y2VjM2RiY2E1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIwVDIxMDYwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5MDMwY2NlNTU0ZDA0NzQzNzdlODFkNDU4OTNlZTQ1MGIxMWQ1MzgzNzA3ZGE4ZTZiYTFiZjNlNzE1NjAzOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.quE6It5-a8XIUt12HwKTqj4JH1UyjsfwNekqJbgqqeI)
> 
> But i like your idea +1 for this i think is similar like my issue [prusa3d/PrusaSlicer#9783](https://github.com/prusa3d/PrusaSlicer/issues/9783)

Thanks for the tip, this is useful, but I would prefer a method of seeing the overhang percentage or angle in a color gradient manner similar to fan speeds, printing speed and flow.

closed 2024-03-27 by github-actions[bot]