OrcaSlicer/OrcaSlicer #3088: Multi color model printed mid air

asychev, 2023-12-11:
### OrcaSlicer Version

1.8.1

### OS version

MacOS

### Printer

P1S

### How to reproduce

Import STL, change color using fill tool, slice

[image]

### Actual results

Part of model printed mid air

### Expected results

Not parts of the model printed mid air

### Project file & Debug log uploads

[double-color-mid-air-issue.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13640782/double-color-mid-air-issue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

blownupp, 2023-12-12:
I think this may be related to #2845

closed 2024-03-19 by github-actions[bot]

asychev, 2024-03-19:
I think this is still an issue