OrcaSlicer/OrcaSlicer #3092: Rotate n degrees for cutting

rkarlsba, 2023-12-12:
Today, if I want to cut somewhere, I can rotate the plane as I wish, but not to a particilar angle. If I want to, say, make a 50x50x10mm cubeoid and slice it in the middle at 37 degrees, there's no way to do this. I guess all it takes, is to add the dialog controls for it - looks like it's all in there, as used in rotate.

closed 2024-03-19 by github-actions[bot]

Klaster1, 2024-10-17:
Second this.

rkarlsba, 2024-10-17:
please add this to orcaslicer!

spooks7er, 2025-05-02:
Second this, this is one feature that bambu slicer still has it better and I still use it because of that. There is no reason setting cut angle and rotating angle to work in different ways.

Currently the circle with the degrees that appears when rotating cutting tools is referencing the original starting position angles and not the plate's angles, which makes it very hard to pinpoint a 90 degree cut on a square piece.