OrcaSlicer/OrcaSlicer #3095: Set direction for Top and Bottom layer skin extrusion direction - like Cura has

Ltek, 2023-12-12:
Please have a setting equal to Cura (separate settings for Top and Bottom) for "Surface skin pattern and line directions"
"The pattern can be set separately to get a different visual effect. If set to lines, the angle of the lines can also be set for more precise control."

Many parts visually looks better ensuring that the line direction is a specific way. Ensuring a specific line direction is not possible with Orca, even when you rotate the model since you cannot control Top and Bottom independently. 

take the example very simple model which you cannot make have the lines go the same direction top and bottom.
[image]

https://github.com/SoftFever/OrcaSlicer/files/12022920/ALP_CoinDoor_Flap_Final_PLA_MatteBlack.zip

Eldenroot, 2023-12-12:
Yes, good idea and I saw a lot of similar request (SS, PS).

Ltek, 2024-03-13:
bueller?

Ltek, 2024-06-14:
not stale... still VERY needed

IanVS, 2024-08-13:
Doh, I was trying to figure out how to do this.  I'm sad that it's not possible.  😢

IanVS, 2024-11-12:
not stale

Ltek, 2024-11-27:
bumpppppp

Ltek, 2025-03-29:
bumper cars

IanVS, 2025-06-29:
Still seems like it would be good to have...

freighttt, 2025-10-07:
also would appreciate this

rowan878, 2026-02-01:
It seems like sometimes when I edit "solid infill direction" it affects the top surface direction, but other times it doesn't have any? I've tried having "align infill direction to model" on and off, and have tried all sorts of different "solid infill direction" angles. For some reason, sometimes the solid infill direction angle works very intuitively in changing the top surface, so if I set my top surface to Monotonic Lines, I can get a perfect outcome. Other times it is unaffected; and what's weirder, I've resliced the same model without changing relevant settings, and it'll stop working :o 

Also, the "align infill direction to model" setting mentions the top surface's orientation, but I skimmed through other settings and couldn't find anything else that mentions it

CautemocSg, 2026-05-07:
@ianalexis - I think the problem is that the top layer cannot be explicitly given a direction, like what I have here is problematic:
https://imgur.com/a/pLuAUpN

I don't seem to have a way to make these layer lines go in the same direction

ianalexis, 2026-05-12:
> @ianalexis on initial user interface inspection it works for the top layer. But I need it to work for both the top layer and the first layer. In my use case, I need both the top and first layers to be at the same angle (in my case 90deg), which is perpendicular to the tips of the guitar picks. This gives me the smoothest surface for the guitar strings when played (no added noise from the filament path. The internal solid layers are then at -45deg and 45deg. 
> 
> I need this enhancement to apply to the first layer as well. As for control UI, either the control you added becomes "Top/Bottom layer direction" which is nice and easy for me. But the needs of others may be different, requiring two controls: Bottom layer direction and Top layer direction. 

For the fist layer you should use the current solid infill direction and this new setting sets the top.