OrcaSlicer/OrcaSlicer #3106: The camera behaves strangely. It rotates around the center of the table, not around the center of the view

FillPavS, 2023-12-13:
### OrcaSlicer Version

1.9.0-dev (make fan percentages a float (#3087) #235)

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

Open the slicer, add any model, move it to the edge, deselect it and rotate the camera.

### Actual results

The camera rotates around the center of the table and not around the model. This happens on both the preparation screen and the preview screen.

video - https://cloud.mail.ru/public/aNCf/94WvwHAqe

### Expected results

The camera should rotate around the model you are looking at. Centered in view

### Project file & Debug log uploads

not required it seems to me

### Checklist of files to include

- [ ] Log file
- [ ] Project file

closed 2024-03-21 by github-actions[bot]

Contomo, 2025-04-22:
As of 2.3.0 beta this is still present..

AussieWarren, 2025-08-01:
Sooo this is intentional?

LJKraus, 2026-08-06:
This is an old topic, but... 

If _Preferences→ Control→ Use free camera_ is checked, rotation is unconstrained.

When the box is unchecked, the Z-axis is locked, and rotation is about the center of the  _selected_ object; if the _plate_ is selected (no object is selected), rotation is about the center of the overall geometry. 

To see this, place three primitives on an empty plate. Group them all near a corner, or space them along an edge, or any other configuration. Rotate rapidly in each configuration while using Top view to see the center.