OrcaSlicer/OrcaSlicer #3107: Spiral Z-Hop can carry nozzle outside of printable area

MiniMook, 2023-12-13:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Additional system information

Ryzen 7 5800X
32GB RAM

### Printer

Voron Trident 300mm

### How to reproduce

1. Load up an object with less than 1-2mm of clearance from the edge of the printable area. I was printing stealthmax parts from https://github.com/nevermore3d/StealthMax on a 300mm Trident
2. Slice using the configuration Printer Settings -> Extruder -> Z Hop Type set to "Spiral"
3. Print file

### Actual results

After a couple hours in you may get an automatic print cancellation with the error similar to:

`Move out of range: 127.384 -0.033 2.662 [14001.338]`

Defining a negative Y value, which is obviously bad.

### Expected results

Printer should complete the print without issues.

I would suggest using `G2` or `G3` for spiral wipes moving towards the center of the bed, rather than always a `G3`

### Project file & Debug log uploads

[Spiral-Z-Hop-Out-of-Bounds.zip](https://github.com/SoftFever/OrcaSlicer/files/13661704/Spiral-Z-Hop-Out-of-Bounds.zip)

The klippy.log may not match this exact project file and this exact gcode file, but the gist of it is the same.

### Checklist of files to include

- [X] Log file
- [X] Project file

closed 2024-03-21 by github-actions[bot]

Technofrikus, 2024-05-19:
I thought I had the same issue but could not reproduce it.

A print failed at the shown layer in the z-hop spiral. With a `Move out of range: -0.019 263.662 23.287 [140557.543] `

[image]
[image]

But only a small part of the print but at the same exact position worked just fine.
[image]

Current best theory is, that the skew correction I use causes this. The part is so close to 0 that this small change from the skew causes the problem.
Assuming you are using klipper, this should solve it. Thanks to Mikl Schmidt for this!

```
[stepper_x]
position_min: -1.5
```

Technofrikus, 2024-05-20:
Confirmed. I added the position_min and ran the same gcode again and it worked without problems.