OrcaSlicer/OrcaSlicer #3108: Z Hop settings

ImminentFate, 2023-12-13:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Printer

Ender 3 S1

### How to reproduce

This seems to be the same issue as #1183, except even the workaround described in that issue doesn't work. 

1. Enable _any_ of the Z Hop settings under the Printer configuration, or even in the filament override as described in #1183
2. Slice model
3. Even attempting to set Z Hop and retractions to ridiculous values (>5mm) still do not work. Using the spiral Z hop, I thought maybe manually enabling arc support would help, but it didn't. 

[image]

### Actual results

Review GCode while panning through layers, there will be no Z Hop at any retractions despite this being set; no vertical travel paths indicating this is occurring either.
You can see it most clearly when ironing is enabled as the nozzle scars the top surface dramatically. 
[image]

### Expected results

Z Hop should be occurring. 

### Project file & Debug log uploads

[FailedZHop.zip](https://github.com/SoftFever/OrcaSlicer/files/13662800/FailedZHop.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ImminentFate, 2023-12-14:
Updated information:
There appears to be a bug that manifests at some point _after_ you save a custom printer profile; leaving it at the default profile (e.g. Creality Ender-3 S1 0.4 nozzle) will allow z hop and retraction settings to work.  Once the bug has occurred, even if you attempt to set all custom values in your profile to match the default, it **still won't work**. You have to make modifications from the stock profile and then overwrite the broken one.  I should have done a diff against the modified configuration file and the stock one to see what actually changed, but unfortunately I didn't before overwriting the broken profile.

closed 2024-03-21 by github-actions[bot]

Novichek-13, 2024-05-22:
The problem is still relevant, I thought that I had something wrong with the position of the z-axis beam or the bed. The nozzle was constantly hitting the model and the supports. Having discovered this topic, I checked the ways of moving the head in ORCA and found that the code for lifting the nozzle does not work when moving. Going into PrusaSlicer 2.7.4 and slicing the model there, I found that everything works as it should. Will there be a solution to this problem in the coming releases? My printer is Anycubic Kobra 2 Pro

XenoChron, 2025-03-23:
This is still a bug/issue as of Marc 23, 2025. I've been having an issue with this with a negative offset and not able to get z hop working on a first layer. Once I removed the offset in Orca I can see the z hop being generated in the Gcode in the slicer.  Is this NOT a bug? It doesn't seem like having the z offset break z hop would be an intentional.

Rico-origin, 2025-03-24:
I also have this problem! I probably destroyed my hot end becouse of this and only realised now, when I installed complete new extruder with hotend upgrade and almost destroyed this one as well. Old heat brake was bent when I dissasembeled the old hotend, and the filament started to push out from the top betwen the heatbrake and heaterblock. I never noticed Z-hop not working problem before, and I assume it was working until now, becouse last succesful printed model was about 250mm tall and it printed great in PLA without artefacts. When first crash occured I had OrcaSlicer 2.2, I created new filament profile for PA-CF (from generic template) and did few small prints with OK results (holow open top box for example - bad choice obviously that is why i did not notice this problem during test prints). Then printed 2 hour long test model for hole clearance. The corners lifted up (I didn't use brims), Z-hop was not working I presume, and hotend crashed and bent the heatbrake. there was a blob of PA-CF on the hotend, when the print finished.
I now upgraded to completley different extruder and hotend combo, flashed new Marlin2 firmware with changes for new extruder/hotend, and updated to OrcaSlicer 2.3.0. I now have this problem with PLA sa well. Luckily I was testing with PLA, as the nozzle was dragging over the printed model and for now it seems the hotend survived. I didn't reslove this issue yet and I will try to adjust the Z offset and report in the evening.

bradyjoh, 2025-12-12:
I was also having this issue, using the latest nightly 2.3.2-dev bd6cfd6. I couldn't get Z hop to work no matter what I tried. I verified by watching the nozzle scrape across whatever I printed, but also enabling the travel view inside the OrcaSlicer Preview. I tried spiral, normal (vertical lift), as well as Slope and Auto.

[image]

I had been using 0 as my values for "Only lift Z above" and "Only lift Z below" as I figured that would mean the feature was disabled and it would always work. Turns out that's not how it works. Now I have it set to everything above 0 and below 300 (my max Z height) and huzzah! I can see the Z hop happening in the slicer now, and it works on the printer now.

I still have the "Reduce infill retraction" Process setting set. I am not using Z offset in my Printer settings. 

Hope this helps someone. I see this issue is closed so I'm not sure if maintainers are picking up that it's still confusing so I'll ping @MxBrnr  and @igiannakas  just to be safe.

daef, 2026-02-16:
> I had been using 0 as my values for "Only lift Z above" and "Only lift Z below" as I
> figured that would mean the feature was disabled and it would always work.
> Turns out that's not how it works.
>
thanks so much! i was looking for my error for ages now, especially since the mouse-over text says it's only enabled for
"positive values"... is 0 positive? dunno... but -1 is not a valid value, so i figured 0 must be off... alas it's not