OrcaSlicer/OrcaSlicer #3110: Max Build Volume Height Not Calculated Correctly After Rotating

niveus, 2023-12-13:
### OrcaSlicer Version

1.8.1

### OS version

Mac OS 14.2

### Additional system information

N/A

### Printer

Creality K1 Max

### How to reproduce

The max printable height check doesn't recalculate after rotating an object. I created a test box with a height of 300 mm (for my K1 Max). When the object is exported from Fusion with the flat face down, Orca will allow slicing. If I export the object rotated so that the max height exceeds 300mm, but then rotate in Orca so that the face is down, Orca doesn't recalculate the max volume height and won't allow it to be sliced.

1. Import box_300 and note that the Slice button is activated
2. Import box_300_skew, plant the box on a small side so the height is 300mm. Note the error around max volume height

[image]

### Actual results

Will not allow slicing

### Expected results

Should allow slicing

### Project file & Debug log uploads

I didn't include the logs since it's so easy to reproduce. LMK if you need the logs. 

[box_300.zip](https://github.com/SoftFever/OrcaSlicer/files/13663229/box_300.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ljohnso16, 2024-01-05:
Same problem on newest 1.9 windows 10. I was able to save project and open project in BBLS and slice just fine.

lukehags007, 2024-01-14:
Same problem

DrewJensen1, 2024-01-19:
Same problem also on the Windows 1.9 version. Potential fix by manually adjusting slicer max build height for printer but unclear of potential effects that could have printer side.

Aurik23, 2024-01-22:
same isse here with BL P1S
- object height: 245mm
- same project possible to slice in BL Studio
[image]

Blogshot, 2024-02-01:
Same problem here. 

It's noteworthy that the prime tower visualizes the issue pretty good, it seems that my ~200mm height gets calculated to around 350mm:
[image]

closed 2024-05-09 by github-actions[bot]