OrcaSlicer/OrcaSlicer #3111: Dragging orcaslicer window is extremely slow

Nuc1eoN, 2023-12-13:
### OrcaSlicer Version

1.8.1

### OS version

Win 10 x64

### Additional system information

AMD RX580, Ryzen 7 3700x

### Printer

does not matter

### How to reproduce

1. Open orcalslicer
2. Drag the window by the title bar

### Actual results

The window will drag VERY slowly.

### Expected results

Window dragging should not be laggy or unresponsive

### Project file & Debug log uploads

not relevant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

subutay, 2023-12-14:
I think so too. Perhaps it's due to the msedgewebview2 component, which is used to display printer web pages. However, this might not be a good idea

RacerX10, 2023-12-14:
I had the same problem under Win 10, recently upgraded to Win 11 and the problem went away.  Might be helpful to troubleshoot.

Nuc1eoN, 2023-12-16:
So looks like it only happens on Win10 so far.

Sortam, 2023-12-19:
I see no such issue. Windows 10, Version 10.0.19044.2965

jum0n, 2023-12-19:
Just installed it and have the same slow window move issue. Microsoft Windows [Version 10.0.19045.3803]

Nuc1eoN, 2023-12-19:
@jum0n I'm also on Win10 22H2 build 19045.3803.

@Sortam interesting. Maybe it doesn't happen on older Win10 versions.

Sortam, 2023-12-20:
@Nuc1eoN Yes, 64 bit (32-bit is a very rare beast these days, I would say :) ). 
I use a laptop with Intel Core i5-1035G4 CPU and its embedded GPU Iris Plus Graphics at screen-native 1920x1080 resolution. 
The laptop has 16 GB of RAM and run from SSD. It's nothing special, and it's obviously not the bleeding edge of technology...

Nuc1eoN, 2023-12-30:
I just tested  this on my laptop which also has recent amd cpu/gpu hardware, but Orcaslicer behaves fine.
My laptop is on Win10 Home 19045.3636, so maybe I could try to update and see if it will break orcaslicer.

Another observation is that the bug only happens for the main Orcaslicer window. So if I open settings or another popup window it will behave fine.

Taskmanager performance tab does not show anything significant.

Iarann, 2024-01-02:
Same behavior here. Win 10 x64, Intel i7-6700k, Nvidia GeForce RTX 2060

Nuc1eoN, 2024-01-02:
> Same behavior here. Win 10 x64, Intel i7-6700k, Nvidia GeForce RTX 2060

Thank you for sharing, so it does not seem related to specific hardware as I have an AMD system.

Would you mind posting your Win10 build version? It should be under Settings -> System -> Info

Iarann, 2024-01-02:
Windows build info:
Edition	Windows 10 Pro
Version	22H2
OS build	19045.3803
Experience	Windows Feature Experience Pack 1000.19053.1000.0

FYI, another piece of info is that Bambu Studio 1.8.2.56 has the exact same behavior on my system.

Nuc1eoN, 2024-01-02:
@AlexanderS Wow thanks the bit about bambu studio being affected is new!

Also it seems that only Win10 22H2 19045.3803 is affected so far. Prior versions work fine.

Also found https://github.com/bambulab/BambuStudio/issues/934 and https://www.reddit.com/r/BambuLab/comments/15uocss/does_anyone_elses_bambu_studio_application_window/

Nuc1eoN, 2024-01-02:
Hi so I have figured out what causes the issue.

It seems that it only happens when your mouse is set to a high polling rate like 1000hz.

Try to set the pollings rate down to the lowest setting and reproduce. I would be very thankful for feedback whether this helps for you too!

PS: Had the polling rate clue from this reddit post: https://www.reddit.com/r/WindowsHelp/comments/p9gwel/slow_motion_windowdragging_bug_on_windows_1011/

Iarann, 2024-01-03:
> It seems that it only happens when your mouse is set to a high polling rate like 1000hz (gaming mice).

@Nuc1eoN Apparently for my system, "high polling rate" is anything about 125hz. Anything higher and I still have "inertia" issues where the screen keeps moving after the mouse stops. But, if I set it for 125hz it eliminates the "inertia", but I still see erratic behavior if I move the screen too quickly.

CypisPiotr, 2024-03-06:
Same problem here (AMD Ryzen 5700G, Win10) - using Logitech G600 mouse,
polling at 250 and more Hz causes the drag on Orca window

elyograg, 2024-03-30:
I'm having a similar problem in Windows 11... but it's with the middle button, dragging the view around. It drags SUPER slow. Rotating the view with the primary mouse button is fast.

CypisPiotr, 2024-03-31:
Still present in 2.0.0

dezenten, 2024-06-09:
Same problem here, maybe I need to upgrade to a faster computer? 
My setup: Windows 10 22H2, Intel I9-12900KS @5Ghz, Nvidia 3090.

jum0n, 2024-06-10:
> Same problem here, maybe I need to upgrade to a faster computer? My setup: Windows 10 22H2, Intel I9-12900KS @5ghz, Nvidia 3090.

You probably need to upgrade to 4090.  The 3090 is known to be slow at melting filament.  :P

Duglum, 2024-08-26:
Can confirm, it's the Mouse polling rate. 1000hz is super slow and sluggish, 125hz works as expected. Bambu Slicer has the same issue while Prusa Slicer works perfectly fine.

Win10 22H2 19045.4780
Orca Slicer v2.2.0-dev nightly from today.
AMD 7950X3D, Nvidia RTX 4090. All drivers are up to date.

This is extremely annoying, hope it will get fixed soon?

MatthewMartinFD, 2024-09-08:
Any update on this? I'm seeing the same behavior.

sgghostrider, 2024-10-14:
I came here from google, just installed OrcaSlicer for the first time and immediately searched it as soon as i saw it. It has been almost a year since the opening of this issue, any plans on fixing this? is this issue even known by the developers?

DoubleStrike, 2024-10-29:
I have a feeling it's the interaction of not just a high mouse rate but also a refresh rate higher than 60 Hz.  I found a similar issue in Blender when dragging the window with the splash screen showing.  This really does need to be elevated in priority, as it is a really frustrating user experience

jum0n, 2024-10-29:
After reducing my mouse polling rate, the issue went away for me.  I'm at 144mhz for display refresh, no issues.

closed 2025-02-05 by github-actions[bot]

Mic0de, 2025-12-26:
I still continue to have this issue. Changing polling rate does not do anything. I only have this issue with this one single application.

ianalexis, 2026-10-29:
Having been affected by this issue myself, it’s a fact that switching to Windows 11 or Linux resolves it. Therefore, I don’t think it’s necessary to add so much logic for a small percentage of users affected in windowed mode when the software is used in full-screen mode most of the time.
I also understand the point about Windows 10 LTS since I used it myself, but the 2032 date refers to an IoT version (which I also used, hahhaa) but if we take that into account, Windows 7 should have been supported until 2024.

radry, 2026-10-29:
@ianalexis 
That's a lazy excuse. Nearly a third of all windows PCs still run Windows 10. Arguing a software is used full screen "most of the time" is a guess based on your personal use case. 

It's not acceptable that a software lags like this in 2026. #16144 aparently fixes this issue, so why continue arguing and not just merge it?

ianalexis, 2026-10-29:
> @ianalexis
> That's a lazy excuse. Nearly a third of all windows PCs still run Windows 10. Arguing a software is used full screen "most of the time" is a guess based on your personal use case.
>
> It's [...]

It's been like this since forever, it has some workarounds and is not a guess is a fact that this affect a small amount of users and just when dragging the window.
Less than 30% of windows users are using W10, and from that just the ones using high refresh rate monitor with high polling rate mouse (and I don't remember if also need to have more than 1 monitor) and not all of them.
As I said, I was one of the users that had this issue and it's nothing that makes the software unusable because it's just when moving the window, even using it in windodow mode do not lag the performance while using it. Worst case scenario is while moving it from one side of the monitor to other or from a monitor to other.

As we said adding code is not a free thing and it can cause new issues, bugs and regresioms while in this case it's not even a high level bug.

If this solution was a thing 2 years ago the history would be different.