OrcaSlicer/OrcaSlicer #3124: Merge with Bambu Studio 1.8.2

Eldenroot, 2023-12-14:
**Is your feature request related to a problem? Please describe.**
Version 1.8.2 was released, brings A1 support and minor fixes https://github.com/bambulab/BambuStudio/releases/tag/v01.08.02.56

**Which printers will be beneficial to this feature**
From this release mainly A1/A1 mini users

**Describe the solution you'd like**
Merge

tsmith35, 2023-12-15:
Yes, Studio 1.8.2 also brings many other changes, including support for P1S printers that were updated from P1P, as well as persistent bed types per-machine, faster slicing, custom filaments, support improvements, improved model visibility when using dark filament, file browsing on P1x/A1x models, connectors for use when splitting objects, etc. [https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-1-8-2](url)

Eldenroot, 2023-12-15:
Well, Orca slicer has features and fixes up to 1.8.1 (included) - including custom filaments, support improvements, model visibility etc.

So 1.8.2 release just contains few commits.

Also Orca slicer includes much more better connectors and splitting object tool (ported from Prusa slicer) than Bambu Lab. 

Please try the latest nightly to test them out :)

tsmith35, 2023-12-15:
Oh definitely. I'm a big fan of OrcaSlicer. It's good stuff!

Eldenroot, 2023-12-15:
Also cool feature from 1.8.2 Bambustudio - you can set the default plate sheet etc.

closed 2023-12-19 by Eldenroot