OrcaSlicer/OrcaSlicer #3131: Paint of brim feature (Prusa slicer - PR port)

Eldenroot, 2023-12-15:
**Is your feature request related to a problem? Please describe.**
(https://github.com/prusa3d/PrusaSlicer/pull/11903)

Really promising PR for one of the most wanted features - painting of brims :) 

Would be great to merge it when it will be ready and done :)

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
Port this PR once be ready.

**Describe alternatives you've considered**
Also extend this to fuzzy skin feature!

arekm, 2024-03-08:
Brim painting would be a really good generic feature (comment, so search engine for issues will hopefully pick up "brim painting" phrase).

For example I like mouse ear brim but auto for it puts these not where I would like to see them (like puts it in some "internal" place etc).

Snuff1eupagus, 2024-04-13:
You have been able to block brims since 2021 as demonstrated by Swiss_Cheese. further development is a waste of time, lets fix things that need fixing effect production.

[image]

[image]

Learn first, then focus on whats important.

rdbischoff, 2024-05-27:
I would love to have brim painting for things like articulated toys. Would be great to have brims in some places and not in others like in between the linked pieces.

Snuff1eupagus, 2024-08-12:
@rdbischoff 

Maybe this will solve the problem for you.

https://www.youtube.com/watch?v=6CRqNk_JUj4

rdbischoff, 2024-08-12:
> @rdbischoff
> 
> Maybe this will solve the problem for you.
> 
> https://www.youtube.com/watch?v=6CRqNk_JUj4

cool. I will check out the video. THANK YOU!!

closed 2024-11-18 by github-actions[bot]