OrcaSlicer/OrcaSlicer #3135: "Create Filament" tab is missing

Eldenroot, 2023-12-15:
### OrcaSlicer Version

1.9.0nightly

### OS version

Win11 64bit

### Printer

A1 mini

### How to reproduce

Filaments settings page - tab "Create filament" is missing. It is present in Bambulab studio.

[image]

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

---

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Eldenroot, 2023-12-15:
When you create a new filament - it is displayed like this in Bambulab studio - I really like this 
[image]

SoftFever, 2023-12-20:
Thanks for reporting the issue.
Fixed.

commit 604851fc (2023-12-20): Fix custom filament tab missing bug
Fixed #3135
closed 2023-12-20 by SoftFever