OrcaSlicer/OrcaSlicer #3136: Hide G-Code viewer (hide/show button)

Eldenroot, 2023-12-15:
**Is your feature request related to a problem? Please describe.**
Add collapsive or show/hide button for G-code viewer

[image]

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
Add an ability to hide/display G-code viewer

**Describe alternatives you've considered**
Now it is sometime annoying and it takes a lot of space.

**Additional context**
---

TechDude23, 2023-12-15:
Hello, 

You can turn it off in the left corner, they have a drop down arrow select preferences and uncheck it.
 
It was removed in the new beta 1.9.0 dev. I don't know why it is not an option anymore, maybe causing problems ?

Eldenroot, 2023-12-15:
Yes, I know. In 1.9dev there is no button anymore

@SoftFever @Noisyfox

darco415, 2023-12-15:
Im on 1.9 dev 1212 and its there

Eldenroot, 2023-12-15:
Where please, cannot see it :(

closed 2023-12-15 by Eldenroot

Noisyfox, 2023-12-16:
Press C to toggle the viewer, or use the view menu.