OrcaSlicer/OrcaSlicer #3137: Orca slicer thumbnail not showing for Qidi xmax 3

ipod9138, 2023-12-15:
Thumbnail unable to be uploaded to Qidi xmax3 printers, using Orca slicer.
Although I and other users have tried different display sizes, I believe its due to some Qidi magic (as on Qidi slicer it uploads, Qidi klipper)
Could this be replicated in an update for Orca slicer?
Thankyou

witchcraftsman, 2024-01-08:
Looks like this issue has been addressed in 1.9.0 alpha

ipod9138, 2024-01-09:
Yes it has, Thankyou to the intelligent ones 🙏🏻👍🏻

closed 2024-01-09 by ipod9138

Dood345, 2024-01-27:
Just in case anyone got a QIDI X-Plus 3 recently, or is just still trying to figure this out I have a very elegant solution that would probably apply to QIDI X-Max 3 as well.
Orca Slicer now has a thumbnail format called "ColPic" which is interpretable by QIDI systems, only problem after that is resolution, I found that a resolution of 380x380 works for all prints.

TLDR: In Orca Slicer select "ColPic" as your format and resolution of 380x380

mohgarr, 2024-07-07:
I changed my settings but still have the issue. Nothing comes up on my qidi x plus 3 screen

nicolalamacchia, 2024-07-25:
I use [this](https://gist.github.com/nicolalamacchia/1f60a4b8e8ca4ac499b1d877ae373c12) script in the post-processing setting: Global Settings -> Others -> [scroll to the bottom] -> Post-processing Scripts -> type the full path of the script you downloaded (make sure that the script is executable).

It works for my X-Plus 3.

Oltre81, 2024-09-09:
@nicolalamacchia Nicola, didn't run for me......scripts working done, but on printer nothing....

Oltre81, 2024-09-30:
Bomba nucleare!

Scusate l'italiano....poi Ve lo tradurrò!

Oggi stampando in pezzo minuscolo mi è uscita l'anteprima! Penso sia questione di scala....

Adesso ci lavoro