OrcaSlicer/OrcaSlicer #3140: Not loading printers

yexele, 2023-12-16:
I have a Comgrow T500, I downloaded the latest version of Orca 1.8.1 from here.  I was unable to load any printer except default.

Win11 pro.  I normally install to D drive instead of C drive so i tried both.  I also tried running as admin.  still nothing.

I downloaded the version from the Comgrow web site V1.8.0-2 beta and it worked just fine no issues except the test file caused the printer to continually say unknown command it keep printing but it came up a lot.

[image]

davideocam, 2023-12-17:
I have just got a T500 too. I can load my other printers (ELEGOO, Crealilty etc.) but not the T500, though it appears as an option.

davideocam, 2023-12-17:
PS Creallity printers seem to load as expected, ELEGOO does not update the preview of bed plate.

yexele, 2023-12-17:
In order for me to use it i had to load the version from Comgrows web site.  It seems others are having issues with the T500 as well on other operating systems.  The mac post got traction and is listed as a bug so it most likely will be fixed in the next release.  Also if your using the T500 you will need to turn off Exclude objects under others.  You constantly get unknown command errors

darkenedforest, 2023-12-20:
This was reported and a hotfix was created. I would expect it to be corrected on version 1.8.2
See more here:
[https://github.com/SoftFever/OrcaSlicer/pull/2999/commits/cc5654b50094aaf6c8b4ab6a2f0ef83e49efbd3e](url)

You can compile a copy of the current code. I was unsuccessful at that, but I was able to hack in the changes. 
You can download: [comgrow_t500_buildplate_texture.png](https://github.com/SoftFever/OrcaSlicer/pull/2999/commits/cc5654b50094aaf6c8b4ab6a2f0ef83e49efbd3e#diff-8f819f365728da8ccf5f6b898aa2f45b77c5bd40a2ec7cc295548b2383aca15f) and [comgrow_t500_buildplate_model.stl](https://github.com/Trist0ne/OrcaSlicer/blob/cc5654b50094aaf6c8b4ab6a2f0ef83e49efbd3e/resources/profiles/Comgrow/comgrow_t500_buildplate_model.stl)

and replace the files located here:
`C:\Program Files\OrcaSlicer\resources\profiles\Comgrow`

Then you can update the JSON file here:
`C:\Users\{your_account_name_here}\AppData\Roaming\OrcaSlicer\system\Comgrow\process\fdm_process_comgrow_common.json`

Add 
`"type": "process",
    "name": "fdm_process_comgrow_common",
    "from": "system",
    "instantiation": "false",`
to the top of the file, so it looks like this:
`{
    "type": "process",
    "name": "fdm_process_comgrow_common",
    "from": "system",
    "instantiation": "false",
    "accel_to_decel_enable": "1",
    "accel_to_decel_factor": "50%",
    "bottom_shell_layers": "2",
...`

and delete  `"from": "",` located down around line 40.

Save the file, then in Orcaslicer, go back to the screen where you add a new printer, make sure your Comgrow T500 choices are checked and click `Confirm`. It will use the modified JSON file to rebuild the printer choices and your printer will show up.

The only down side is that when you close and reopen, the JSON file will get rebuilt, and you will have to modify the JSON file again. But it should at least get you by until 1.8.2 rolls out.

yexele, 2023-12-20:
Thank you so much I'll just wait until it rolls out the old one is working well enough for what I currently am doing.  Ps great work I'm liking what I'm seeing if you ever need any help let me know

https://github.com/SoftFever/OrcaSlicer/pull/2999/commits/cc5654b50094aaf6c8b4ab6a2f0ef83e49efbd3e

closed 2024-03-27 by github-actions[bot]