OrcaSlicer/OrcaSlicer #3144: Small perimeters and top layers noticable worse compared to PrusaSlicer

Boerny41, 2023-12-16:
### OrcaSlicer Version

1.8.0 / 1.8.1

### OS version

Windows 10

### Printer

Anycubic Mega SE

### How to reproduce

1. Download PursaSlicer
2. Copy every setting
3. Print same STL

### Actual results

First of all, I feel like this isn't the best place to post this, but I don't know what else to do. I've been trying to get OrcaSlicer to work for 3 weeks now, but it keeps showing the same problems and I'm at the point where I think it might actually be a problem with the version.

I've noticed that small perimeters and top layers are always noticeably worse compared to PrusaSlicer, even when using the exact same settings in both slicers. Below I'll post some pictures with Orca on the left and Prusa on the right. I now have a bin full of test prints, all printed with different settings, but all showing the same differences between Prusa and Orca, e.g. internal bridging slowed down for testing in both slicers, same stl sliced in both and printed.

I asked on reddit, but people there were about as helpful as you'd expect, "obviously under extrusion, pump those numbers up". While I understand that the rectangles look like they've been under-extruded, I've done countless test prints with all the usual settings and the problems keep reappearing, but only for OrcaSlicer.

I have also verified for myself that it is not the normal print settings such as flow, speed, thickness or similar, as the same settings in Prusa give the expected results.

[image]
[image]
[image]
[image]
[image]
[image]

### Expected results

Same print quality since Orca is a fork of Prusa.

### Project file & Debug log uploads

[TestPrintScrewholes.zip](https://github.com/SoftFever/OrcaSlicer/files/13692605/TestPrintScrewholes.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

dartrax, 2024-03-16:
Is this still an issue with v2.0 beta?

PS: I think it is the right place to report that issue here.

Boerny41, 2024-03-16:
Just yesterday I wanted to know the same thing and made some more prints. They look a lot better, but I'm not entirely convinced that the problem is solved. 

It was always pretty hard to reproduce, and those little blocks with a hole and nut insert just failed a lot, so I used them as a test. Also, I don't think it was an actual bug in the slicer. I made two posts on reddit, asked twice on Orca Discord, and made this post here. It was rarely answered and no one has ever had similar issues. If it was a bug, I'm sure someone with the same problem would have seen my posts.

On the picture you can see the old prints from December in glitter black and the new ones from yesterday in black.

[image]

dartrax, 2024-03-16:
Thanks for the update.
The new ones look good, but it‘s not clear if that‘s the effect of a bugfix in orca or the use of another filament.
You could try to print in gitter black or use the new black filament with the old version to see if that produces the underextrusion also with your black filament.

closed 2024-06-24 by github-actions[bot]