OrcaSlicer/OrcaSlicer #3146: Reverse on odd causes weird extrusion inconsistencies

Parozzz, 2023-12-16:
### OrcaSlicer Version

1.8.1

### OS version

Windows 10

### Printer

Voron 2.4

### How to reproduce

1. Enable "Reverse on odd"
2. Print an Orca Cube (Very easy to see the problem)

### Actual results

After enabling the "Reverse on odd" feature there is a very notable understrusion for the layer that are reversed.
After many test (I couldn't figure out what was the problem since with a similar profile on SuperSlicer i didn't have this problem) i concluded that this is more visible with small objects hence i haven't noticed before.

As you can see from the pictures, one have visible indented layer (One good and one bad) the other is fine.

[image]
[image]

### Expected results

Maintain print quality the same with and without "Reverse on odd" option enabled.

### Project file & Debug log uploads

For model, load primitive "Orca cube" model.

[Profiles.zip](https://github.com/SoftFever/OrcaSlicer/files/13692730/Profiles.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-12-16:
That's expected.

Parozzz, 2023-12-16:
> That's expected. There is simply no way to have consistency extrusion when direction is alternated because no machine is perfectly built. In 1.9 we will introduce a feature that only alternates direction of inner walls so the surface quality won't be affected.

Oh, something i never figured out before. Thanks for the insight. Closing this since is expected.

closed 2023-12-16 by Parozzz