OrcaSlicer/OrcaSlicer #3150: Can't get rid of stringing

EkiciLP, 2023-12-16:
### OrcaSlicer Version

1.8.1

### OS version

Windows 10

### Printer

Anycubic Vyper with Direct Extruder mod

### How to reproduce

1. Slice some Object
2. Print it
3. See the Strings

### Actual results

I see many Strings and even Blobs which I don't see if I slice with Superslicer

### Expected results

No stringing. I already tried solutions of other posts like using Spiral Z-Hop which significantly improved the strings but they are still there and sometimes my Parts don't fit together because of blobs. I don't know if I just didn't copy the settings correctly but it seems right to me. I've been troubleshooting with different speeds and lengths but the results are always almost identical. With Superslicer there are other issues and in my opinion Orca Slicer is just way better, partly because it is actually maintained.

I will attach both my Orcaslicer and Superslicer profiles so you can check that I haven't done some dumb mistake.

### Project file & Debug log uploads

[Configs.zip](https://github.com/SoftFever/OrcaSlicer/files/13693610/Configs.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

oscillatingdecay, 2023-12-20:
I am also experiencing this. I think there is something wrong with "retraction" was a whole. I am experiencing the string but if you look at the first layer, the seam is blobbing and not retracting. As soon as I imported the settings and sliced the model in Prusaslicer, the blobs on the first layer went away where the reatraction would have occurred.

NeverUsedID, 2023-12-22:
The reason is "wipe while retraction" on my side.  I also can't see any retraction g-code in preview or in exported gcode file. So only Wipe without retraction is happening.

Example1: "retraction on layer change" enabled, but no wipe:

[image]

Example2: Wipe enabled
[image]

I get the stringing exactly on the wipe positions allong these travel lines:

[image]

Keep in mind that i'am really new to 3d-printing.

kadisonm, 2024-01-09:
I'm experiencing the same problem and it looks like other people have been as well. 

https://www.reddit.com/r/FixMyPrint/comments/1764ewy/stringing_after_switching_to_orca_slicer/

It seems like this [user](https://www.reddit.com/r/FixMyPrint/comments/1764ewy/comment/k9quany/?utm_source=share&utm_medium=web2x&context=3) commented something similar to @NeverUsedID, so maybe this has something to do with the stringing issues?

I've done a couple of things to prevent stringing but I'm still getting it especially on my organic supports. It isn't horrible but it is way more than what I get on Bambu Studio. If it's relevant I am using a Bambu Lab P1S with Sunlu PLA+.

What I've tried:
- Increasing retraction length (.9mm)
- Spiral Z hop
- Retract on layer change
- Setting retraction speed to the same as Bambu Studio
- Wipe while retracting
- Wipe distance to 2mm and retract amount before wipe to 10%
- Turning reduce infill retraction off.

user6508, 2024-01-11:
The same issue I have with orca and prusa.
Cura gives more clearer result.

faberyx, 2024-01-20:
have the same issue too, tried a lot of different settings but I still get it, it doesn't happen at all on Cura

Naninani, 2024-01-26:
I think the issue is with "Wipe while retracting" not working at all.

**"Wipe while retracting" disabled, "Retract amount before wipe" set to 100% or any other %**
[image]
[image]

**"Wipe while retracting" enabled, "Retract amount before wipe" set to 50%**
[image]

**"Wipe while retracting" enabled, "Retract amount before wipe" set to 0%**
[image]

@SoftFever, looks like `Wipe while retracting` is not working.

yonkiman, 2024-02-08:
I'm seeing two problems related to this.  The first is that there is a lot of long (>100mm) travel without retracting.  So there's a lot of oozing out of my 0.6mm nozzle.

The second is that rotating the part in Z by 2 degrees (the only way I can put the seam close to where I want it ) radically changes the slicing, so there is a _ton_ more travel (again, with no retractions) than the identical but non-rotated part in the upper right.
[image]
[Travel Changes.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/14215335/Travel.Changes.3mf.zip)

Edzi11, 2024-02-16:
Ive been having fuzz and stringing as well that disappears when using prusa slicer. same settings as well.

sixerstyle, 2024-02-29:
Same here, you can't get stringing under control.
no problem with cura or prusa slicer

oscillatingdecay, 2024-04-06:
Wipe on retracting is not occurring for me.
[image]

grrminator, 2024-04-16:
Its very frustrating. I have multiple printers that were on cura.

As soon as I move to orca, which is much better, this stringing will NOT go away. I get that its fuzz and mostly can just be burned off, I shouldnt have to get a butane torch every time my printer finishes something.

sixerstyle, 2024-04-29:
Since version 2.0.0 the stringing problems have disappeared for me

staal54a, 2024-07-18:
> I think the issue is with "Wipe while retracting" not working at all.
> 
> **"Wipe while retracting" disabled, "Retract amount before wipe" set to 100% or any other %** <img alt="image" width="1406" src="https://private-user-images.githubusercontent.com/1681582/300076165-63c2b38e-42c8-4092-a79e-a62ab8c87a06.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzIxMDcsIm5iZiI6MTcyMTI3MTgwNywicGF0aCI6Ii8xNjgxNTgyLzMwMDA3NjE2NS02M2MyYjM4ZS00MmM4LTQwOTItYTc5ZS1hNjJhYjhjODdhMDYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MThUMDMwMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjhiOTIyOGYxNjg2OWEzOWYyNzI1ZTQ0NTM5ODZmODQ3MzMyYzU5ZmUwNTE0OWRmMTk4NzAyZjdjMDdjMWIxNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yqlFQaEKx1Sa9MlxHwQIWaQaegyN0PjywafiOARr5Lc"> <img alt="image" width="1254" src="https://private-user-images.githubusercontent.com/1681582/300077255-d403fd11-5ac7-4d38-822d-bedb6853ab04.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzIxMDcsIm5iZiI6MTcyMTI3MTgwNywicGF0aCI6Ii8xNjgxNTgyLzMwMDA3NzI1NS1kNDAzZmQxMS01YWM3LTRkMzgtODIyZC1iZWRiNjg1M2FiMDQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MThUMDMwMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODIyNGY1MGI5NjI5YWYyM2RhYjdlMzRmNTEzOWM5NGQwNmNjZmYyNzBhMTY2OGYwYWNjMTEwMWE0YjNmYTc4MiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.FyyKAqSl0yDVYh5F6NKC1IIDmadIpg98KNqLzU_W0I4">
> 
> **"Wipe while retracting" enabled, "Retract amount before wipe" set to 50%** <img alt="image" width="1411" src="https://private-user-images.githubusercontent.com/1681582/300076722-c4c1a55c-9325-4424-af5c-5778bb326e5a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzIxMDcsIm5iZiI6MTcyMTI3MTgwNywicGF0aCI6Ii8xNjgxNTgyLzMwMDA3NjcyMi1jNGMxYTU1Yy05MzI1LTQ0MjQtYWY1Yy01Nzc4YmIzMjZlNWEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MThUMDMwMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YWIwZmZlNTZkZGJjY2JhZWUwMzAxMTdlOTE3Y2Y2OGY5YmM4MjNjNWQ2MTQzYzdjZmI4NjA4NzI5ZmIxN2RjZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.jptmoHu3sDN62Bw44eUvHnvewtFKZw6yc43PbMYIVfM">
> 
> **"Wipe while retracting" enabled, "Retract amount before wipe" set to 0%** <img alt="image" width="1314" src="https://private-user-images.githubusercontent.com/1681582/300077121-1eb9e615-f93e-4bc5-8444-bfbf75ffe36e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzIxMDcsIm5iZiI6MTcyMTI3MTgwNywicGF0aCI6Ii8xNjgxNTgyLzMwMDA3NzEyMS0xZWI5ZTYxNS1mOTNlLTRiYzUtODQ0NC1iZmJmNzVmZmUzNmUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MThUMDMwMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjI4N2Y4MDljNTE1NzhmMTRjYTA1NjljNjk1MzhjYTlkNDk5NWJlZWQyOWZlOTliMDY5MDZlNThiYjcxZTI0YyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.I6_pvV_sHrtnzq7HLIZvn0_owPmHa1EYxTqWzdblWc0">
> 
> Also for whatever reason the `Z hop type` on the inner wall after the wipe is `Normal` instead of `Spiral`. <img alt="image" width="1265" src="https://private-user-images.githubusercontent.com/1681582/300078783-58613fb1-4a68-47ca-8227-086b919b6111.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzIxMDcsIm5iZiI6MTcyMTI3MTgwNywicGF0aCI6Ii8xNjgxNTgyLzMwMDA3ODc4My01ODYxM2ZiMS00YTY4LTQ3Y2EtODIyNy0wODZiOTE5YjYxMTEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MThUMDMwMzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjY2N2MwNTk5ZTViZDc5NzlkOWUwNzFkMTAyNjMwNzc3MmJlYTM1YTIxYzQ5Y2Y0YjliMDVkMjM4ZWZjNzRkMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.FYoD4GSLSFyxA7Cj09Rwf9-Umjpw-sISChumuKCWEvA">
> 
> @SoftFever, looks like `Wipe while retracting` is not working.

I can't seem to load some of your images but at least in that last image, if you zoom all the way in on where those retractions to z-hops are occurring you can see the yellow line indicating the wipe. It is a much smaller line than any of the walls (it's about the same size as the travel/z-hop lines versus teh printing wall lines)

staal54a, 2024-07-18:
> Wipe on retracting is not occurring for me. ![Screenshot 2024-04-05 at 6 42 52 PM](https://private-user-images.githubusercontent.com/141880001/320166587-97813aff-9283-460a-887d-11e93fc008bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzIxMDcsIm5iZiI6MTcyMTI3MTgwNywicGF0aCI6Ii8xNDE4ODAwMDEvMzIwMTY2NTg3LTk3ODEzYWZmLTkyODMtNDYwYS04ODdkLTExZTkzZmMwMDhiYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzE4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcxOFQwMzAzMjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZDliZTM3Y2U3YzRlNmUzZGY1YTg2NDZmMzBlNDIxNDM1ZDcxYTk2OWIxNmEzMzBjMzVkYzg4NTRlNTAxN2IxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lc4L_HGlDx843QksaqOWdKC5pOfDSL_jxYEF44O2_Ww)

And similar for your issue, if you zoom in on the rightmost inner wall, there is a small yellow line above the inner wall line which indicates a wipe is occurring. The interesting part is why it's not occurring on all those retractions but also tough to gather without seeing the retraction settings and such.

I personally haven't had an issue getting wipe while retracting to work in any version of Orca I've used (since pre 1.0) but the main issue I'm seeing is that the retract amount before wipe percentage is not truly honored. Basically it seems that the wipe distance and speed is used to calculate the amount of retraction that occurs during the wipe, then any additional retraction required occurs before the wipe.

ForsakenRei, 2024-08-09:
I'm also having stringing issue. Temp tower printed and chose a best temp for my filament(190C surprisingly) and printed a retraction test which is perfect, but when printing string tower or my normal parts I got tons of strings. When looking at it actually prints the strings tower, it seems my spiral z hop only happens on 1 of the 2 towers towards the tip so one of them will have a lot of strings. I don't know why it only do z hop on that one tower.

DearTanker, 2024-09-15:
Because I use the Creality K1C, I have never tried Bambu Studio before.

But, recently I found that Bambu Studio can actually support non-Bambu 3D printers, so I tried it with the string hell model with it. I didn't expect that the result was very perfect. I have ever seen that good result before. After many days with countless comparisons and tweak with Orca, I finally discovered a detail that might be causing this problem, and you can't set it through Orca.

[image]
[left: Bambu Studio][right: OrcaSlicer]

Bambu Studio will retraction again before traveling after retracting and wiping, while Orca will not do it no matter how I adjust the settings. I think this is the root cause of the strings problem.

staal54a, 2024-10-30:
> Because I use the Creality K1C, I have never tried Bambu Studio before.
> 
> But, recently I found that Bambu Studio can actually support non-Bambu 3D printers, so I tried it with the string hell model with it. I didn't expect that the result was very perfect. I have ever seen that good result before. After many days with countless comparisons and tweak with Orca, I finally discovered a detail that might be causing this problem, and you can't set it through Orca.
> 
> <img alt="2024-09-15 162125" width="1917" src="https://private-user-images.githubusercontent.com/1122669/367569277-28d9eafa-9faa-479e-9ba1-74a13fd00014.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyNTE2ODUsIm5iZiI6MTczMDI1MTM4NSwicGF0aCI6Ii8xMTIyNjY5LzM2NzU2OTI3Ny0yOGQ5ZWFmYS05ZmFhLTQ3OWUtOWJhMS03NGExM2ZkMDAwMTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMzBUMDEyMzA1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NGUyMzQxMWRmMjQ4M2EwNDQyM2EzZWIwYjg4MzY3MzU1MTQxZTJhNjNkMzZiZDM2YWJjY2NhNjQ0N2M2OTg0NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.FnZ8Nl8aPZ6hTWEQvPA8jRDcEuPSUQAooykjCeuvZmU"> [left: Bambu Studio][right: OrcaSlicer]
> Bambu Studio will retraction again before traveling after retracting and wiping, while Orca will not do it no matter how I adjust the settings. I think this is the root cause of the strings problem.

I'm pretty sure what you are seeing is the fact that Bambu Studio will respect the 0% retract before wipe value but Orca does not. So my guess as to what you are seeing is the following (which I was able to recreate in Bambu Studio and Orca Slicer):

- In Bambu Studio, with 0% retract before wipe, the wipe movement and initial retraction happen at the same time. If at the end of the wipe, the retraction length has not been reached, Bambu Studio inserts a retraction for the remaining amount. If you have a model with a bunch of retract and wipes on straight lines, the remaining retraction amount should be pretty consistent.
- In Orca Slicer, with 0% retract before wipe, the slicer will calculate the retraction length it would NOT be able to do within the requested wipe distance. It then attempts to retract that before it starts the wipe (unless the value is 0 in which case it skips this). So instead of a line finishing and the wipe starting immediately, there is really a (potential) pause while it does the initial retraction, then the wipe and retraction starts.

So I'm wondering if the wipe not starting immediately causes any issues since the nozzle would just be sitting there, albeit for a potentially really short period of time. I may play around with some different parameters (wipe speed and distance namely) to see if that improves the stringing issues I'm seeing myself.

Milliw, 2024-12-20:
Just want to add my experience here, too.

Was using Cura since I got my first 3D printer. After gaining more knowledge I switched to OrcaSlicer. I find it far the better program. But stringing got more of an issue at sudden. Did excessive tests with a simple, small stringing tower stl. Tried different settings. Then printed same model with Cura and boom - stringing not an issue! So I switched back to Cura for now sadly.

But I do want to mention, that I tried PrucaSlicer as well and I got same results as with OrcaSlicer. Bummer.

Anycubic Mega S

Greetings

MajorKingston, 2025-02-23:
Hey Guys! 
I had the same Problem on my Anycubic Kobra Plus with the stringing. The retraction test didnt have any differences whatsoever.

For me it was a Problem with the machine G-Code und the setting for Relative E.

Before the Fix, the Values for the extrusion in the G-Code were always positive float Numbers.
Afterwards they were negative an it was easy too see the steps in the retraction test G-Code.

I had to change this value in the machine G-Code, which sets back the extrusion value to 0 before the layer change, this value was commented out.

[image]

Afterward i set my relative e  to on.

[image]

That fixed the stringing problem for me. Maybe its a solution for at least some of you.

closed 2025-06-01 by github-actions[bot]