OrcaSlicer/OrcaSlicer #3152: Filament settings not saved after restart.

poolboy97, 2023-12-16:
### OrcaSlicer Version

1.8.0 and 1.8.1

### OS version

windows 11

### Printer

cr10 v3/klipper

### How to reproduce

1. select printer/setup custom
2. create custom filament profile for pla, save as user preset
3. run calibrations and try to save new value, or a does not let you select "user preset" to save new values. 

### Actual results

User preset not available, new values saved until you close orca and start back up. 
[image]
  

### Expected results

User preset for filament profile should save. after start up filament profile made/saved should be available. Once you restart Orca filament profile returns to default and the saved preset is not there. I have tried to export profiles and a folder is created and says it contains the presets but when you import them back to Orca message says no files to import.

[image]

### Project file & Debug log uploads

[debug_network_Fri_Dec_15_15_19_05.zip](https://github.com/SoftFever/OrcaSlicer/files/13693971/debug_network_Fri_Dec_15_15_19_05.zip)
[crash_Sat_Dec_02_23_33_45_0.zip](https://github.com/SoftFever/OrcaSlicer/files/13693972/crash_Sat_Dec_02_23_33_45_0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

poolboy97, 2023-12-16:
I want to use orca as my primary slicer but this issue is very annoying as i have to go back through every time i start orca remake and calibrate all if my settings which is a major pain in the tail. I generally print with various materials which makes it even worse to setup for each every time i change materials.

matt8227, 2023-12-16:
To add some information as I am facing the same issue:
the filament json files get saved in appdata, but even if you try to import them manually OrcaSlicer says there are 0 files imported.

I also noticed that VulpinePrints has found a possible dirty fix for a similar issue, described in issue #1182

SoftFever, 2023-12-17:
Have you added filament before?
Currently there is an issue if you edit and save based on the "default filament"

matt8227, 2023-12-17:
No, the first version I installed is 1.8.1, and since now I just edited the default filament every time since I am a complete noob at 3d printing lol.
I just added the generic filament presets, edited the pla one and it seems like it saved, thank you.

poolboy97, 2023-12-17:
I have tried creating my own profile, it doesn't allow me to. I have selected generic filament profiles and have tried to edit/save them but when i go to drop down to select it only gives "default" as an option. So pretty much the only thing i can do is to edit the default and try to save "which it does not do" each time.

matt8227, 2023-12-17:
What I did is I added a new generic klipper printer, which also added system presets for generic materials. I edited the PLA one since that is what I am using and then saved it, and it works for me now.

poolboy97, 2023-12-17:
Will give that a shot then.

poolboy97, 2023-12-17:
Definitely appreciate it. Hopefully this will get fixed in the near upcoming updates. But the work around will work for now.

closed 2024-03-24 by github-actions[bot]