OrcaSlicer/OrcaSlicer #3154: Ellis PA calibration has dangerously bugged extrusion

Habag111, 2023-12-17:
### OrcaSlicer Version

1.8.1

### OS version

Windows 10

### Printer

Ender 3 v2 with klipper

### How to reproduce

Use pressure advance calibration pattern (I used start 0, end 0,46, step 0,02) and "use relative E distances" disabled. with it enabled the issue still arises, but is less noticeable

### Actual results

at the end of the first layer when the pattern ends and the cube next to it is being printed, a "G92 0" command isn't issued (which results in the filament being retracted a long way, in my case 480 units). Then, after the cube is finished, a "G92 0" is issued, BUT the E distance that was used in the pattern before the cube is still used, which results (in my case) in a request for 480 units of E, which is impossible and makes the extruder skip.

### Expected results

Generated Gcode has proper extrusion commands.

### Project file & Debug log uploads

[PA_bug.zip](https://github.com/SoftFever/OrcaSlicer/files/13694307/PA_bug.zip)

I included both the saved orca .3mf file, and the Gcode. the problem can be seen around the 2150 - 2250 lines of Gcode (screenshots attached)
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

Habag111, 2023-12-27:
The issue is still present in 1.9.0 release. Screenshots with explanation attached.

[image]

[image]

igiannakas, 2023-12-27:
Hi, I have reproduced the issue. I'll try and fix it soon.

igiannakas, 2023-12-27:
Can you try this build here: https://github.com/SoftFever/OrcaSlicer/pull/3314

Habag111, 2023-12-27:
I haven't tried actually printing it, but looking at the generated gcode, it seems you fixed it! Thank you so much! 

Should I mark this issue as closed? I'm pretty new to github, so I'm not sure if that's whats appropriate.

igiannakas, 2023-12-27:
yes please. However there is one more issue with it that is not related to this bug which I'm trying to fix. You're not encountering it as you have turned off the retract on layer change option

Habag111, 2024-02-20:
Reopening because the bug is not completely fixed. 

The issue still arises, but now only on the last layer of the cube. This happens at line 3668 of the Gcode I attached as a zip. Screenshot and project file also attached. I'm using the official 1.9.1 version on win10. 

[image]

[PA_bug_orca_1_9_1.zip](https://github.com/SoftFever/OrcaSlicer/files/14341699/PA_bug_orca_1_9_1.zip)

Habag111, 2024-02-25:
@igiannakas Sorry for tagging but don't want this to get lost, especially since my nozzle clogged because of the bug and I don't want anyone else to face the same problem

closed 2024-06-02 by github-actions[bot]