OrcaSlicer/OrcaSlicer #3157: Consistent Crash

triumphandblunder, 2023-12-17:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11 Pro build 10.0.22631

### Additional system information

CPU: Ryzen 9 3900x
Graphics: RTX 2080 Super Driver Version 546.17
Ram: 64gigs installed

### Printer

Bambu Lab P1S

### How to reproduce

Honestly hard to reproduce. Its up, I am using and it crashes. Sometimes during use sometimes I come back and its gone. I got this recently in the logs.

Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x60 
Context :00000059FF52CFC0 
ContextFlag : 0x10005f, EFlags: 0x10246 
Fault address:  0x68D4F8A2 0x1:0x101E8A2 C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:
7ffe68d4f8a2:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffe68cf0bd9:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffe68cefed7:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffe68b4ad59:cereal::detail::StaticObject<cereal::detail::PolymorphicVirtualCaster<Slic3r::ObjectWithTimestamp,Slic3r::LayerHeightProfile> >::instantiate [Unknown file][-1]
7ffe68b4a25e:cereal::detail::StaticObject<cereal::detail::PolymorphicVirtualCaster<Slic3r::ObjectWithTimestamp,Slic3r::LayerHeightProfile> >::instantiate [Unknown file][-1]
7ffe695bdbd4:hid_write [Unknown file][-1]
7ffe695b96bf:hid_write [Unknown file][-1]
7ffe695b9aaa:hid_write [Unknown file][-1]
7ffe695b9d5f:hid_write [Unknown file][-1]
7ffe695b95eb:hid_write [Unknown file][-1]
7ffe695b9a1a:hid_write [Unknown file][-1]
7ffe696354ac:hid_write [Unknown file][-1]
7ffe69636395:hid_write [Unknown file][-1]
7ffef36a8241:DispatchMessageW [Unknown file][-1]
7ffef36a7d01:DispatchMessageW [Unknown file][-1]
7ffef36a8b6d:IsDialogMessageW [Unknown file][-1]
7ffe6946ef80:hid_write [Unknown file][-1]
7ffe6946f0c2:hid_write [Unknown file][-1]
7ffe6946e9ee:hid_write [Unknown file][-1]
7ffe695f9f03:hid_write [Unknown file][-1]
7ffe695fa14a:hid_write [Unknown file][-1]
7ffe695bebc0:hid_write [Unknown file][-1]
7ffe695c28a7:hid_write [Unknown file][-1]
7ffe696278b8:hid_write [Unknown file][-1]
7ffe6860554e:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffe67e06b69:Unknown [Unknown file][-1]
7ffe67e23e7d:orcaslicer_main [Unknown file][-1]
7ff634f75945:Unknown [Unknown file][-1]
7ff634f75e06:Unknown [Unknown file][-1]
7ffef296257d:BaseThreadInitThunk [Unknown file][-1]
7ffef3eaaa58:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x68D4F8A2 0x1:0x101E8A2 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x68CF0BD9 0x1:0xFBFBD9 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x68CEFED7 0x1:0xFBEED7 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x68B4AD59 0x1:0xE19D59 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x68B4A25E 0x1:0xE1925E C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695BDBD4 0x1:0x188CBD4 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695B96BF 0x1:0x18886BF C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695B9AAA 0x1:0x1888AAA C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695B9D5F 0x1:0x1888D5F C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695B95EB 0x1:0x18885EB C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695B9A1A 0x1:0x1888A1A C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x696354AC 0x1:0x19044AC C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x69636395 0x1:0x1905395 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xF36A8241 0x1:0x17241 C:\WINDOWS\System32\USER32.dll
0xF36A7D01 0x1:0x16D01 C:\WINDOWS\System32\USER32.dll
0xF36A8B6D 0x1:0x17B6D C:\WINDOWS\System32\USER32.dll
0x6946EF80 0x1:0x173DF80 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x6946F0C2 0x1:0x173E0C2 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x6946E9EE 0x1:0x173D9EE C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695F9F03 0x1:0x18C8F03 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695FA14A 0x1:0x18C914A C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695BEBC0 0x1:0x188DBC0 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x695C28A7 0x1:0x18918A7 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x696278B8 0x1:0x18F68B8 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x6860554E 0x1:0x8D454E C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x67E06B69 0x1:0xD5B69 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x67E23E7D 0x1:0xF2E7D C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x34F75945 0x1:0x4945 C:\Program Files\OrcaSlicer\orca-slicer.exe
0x34F75E06 0x1:0x4E06 C:\Program Files\OrcaSlicer\orca-slicer.exe
0xF296257D 0x1:0x1157D C:\WINDOWS\System32\KERNEL32.DLL
0xF3EAAA58 0x1:0x59A58 C:\WINDOWS\SYSTEM32\ntdll.dll

### Actual results

Crashes

### Expected results

Shouldnt crash

### Project file & Debug log uploads

f

### Checklist of files to include

- [ ] Log file
- [ ] Project file

xorinzor, 2023-12-17:
For me it happens very often while going to the device tab.

Version: 1.8.1
Printer: Bambu X1C

```
Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x18 
Context :000000147ABCD440 
ContextFlag : 0x10005f, EFlags: 0x10246 
Fault address:  0x77E5F45B 0x1:0xAE45B C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:
7ffd77e5f45b:Unknown [Unknown file][-1]
7ffd78df162e:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffd78df1f2a:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffd78def86b:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffd78d6ffa9:cereal::detail::StaticObject<cereal::detail::Versions>::lock [Unknown file][-1]
7ffd78bcad59:cereal::detail::StaticObject<cereal::detail::PolymorphicVirtualCaster<Slic3r::ObjectWithTimestamp,Slic3r::LayerHeightProfile> >::instantiate [Unknown file][-1]
7ffd78bca25e:cereal::detail::StaticObject<cereal::detail::PolymorphicVirtualCaster<Slic3r::ObjectWithTimestamp,Slic3r::LayerHeightProfile> >::instantiate [Unknown file][-1]
7ffd7963dbd4:hid_write [Unknown file][-1]
7ffd796396bf:hid_write [Unknown file][-1]
7ffd79639aaa:hid_write [Unknown file][-1]
7ffd79639d5f:hid_write [Unknown file][-1]
7ffd796395eb:hid_write [Unknown file][-1]
7ffd796398ab:hid_write [Unknown file][-1]
7ffd7963f23b:hid_write [Unknown file][-1]
7ffd79679ef4:hid_write [Unknown file][-1]
7ffd7967a14a:hid_write [Unknown file][-1]
7ffd7963ebc0:hid_write [Unknown file][-1]
7ffd796428a7:hid_write [Unknown file][-1]
7ffd796a78b8:hid_write [Unknown file][-1]
7ffd7868554e:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffd77e86b69:Unknown [Unknown file][-1]
7ffd77ea3e7d:orcaslicer_main [Unknown file][-1]
7ff78ad75945:Unknown [Unknown file][-1]
7ff78ad75e06:Unknown [Unknown file][-1]
7ffde45a257d:BaseThreadInitThunk [Unknown file][-1]
7ffde596aa58:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x77E5F45B 0x1:0xAE45B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x78DF162E 0x1:0x104062E C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x78DF1F2A 0x1:0x1040F2A C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x78DEF86B 0x1:0x103E86B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x78D6FFA9 0x1:0xFBEFA9 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x78BCAD59 0x1:0xE19D59 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x78BCA25E 0x1:0xE1925E C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x7963DBD4 0x1:0x188CBD4 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x796396BF 0x1:0x18886BF C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x79639AAA 0x1:0x1888AAA C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x79639D5F 0x1:0x1888D5F C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x796395EB 0x1:0x18885EB C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x796398AB 0x1:0x18888AB C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x7963F23B 0x1:0x188E23B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x79679EF4 0x1:0x18C8EF4 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x7967A14A 0x1:0x18C914A C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x7963EBC0 0x1:0x188DBC0 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x796428A7 0x1:0x18918A7 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x796A78B8 0x1:0x18F68B8 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x7868554E 0x1:0x8D454E C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x77E86B69 0x1:0xD5B69 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x77EA3E7D 0x1:0xF2E7D C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0x8AD75945 0x1:0x4945 C:\Program Files\OrcaSlicer\orca-slicer.exe
0x8AD75E06 0x1:0x4E06 C:\Program Files\OrcaSlicer\orca-slicer.exe
0xE45A257D 0x1:0x1157D C:\Windows\System32\KERNEL32.dll
0xE596AA58 0x1:0x59A58 C:\Windows\SYSTEM32\ntdll.dll
```

triumphandblunder, 2023-12-17:
Mine does also happen at that point. But even happens when its on the
preview and prepare tab.

triumphandblunder, 2023-12-20:
God I wish someone would figure out whats up with this. Bambu Studio doesnt do it.

xorinzor, 2023-12-20:
There's no need for comments like that, they don't help with the issue and are irrelevant.
It's open source software, people are doing this for free. You cannot expect them to be working on this full-time.

triumphandblunder, 2023-12-20:
Comments saying that Bambu Studio doesnt crash consistently as Orcaslicer does? I mean....that's facts

xorinzor, 2023-12-20:
That bit, sure. But comments like "God I wish someone would figure out whats up with this" after only 4 days puts a lot of pressure on developers.

triumphandblunder, 2023-12-20:
No it doesnt, they do what they can when they do. I was stating the facts. I have multiple crashes a day, with no rhyme or reason, I have a machine that can handle it and bambu studio's without any issues.

What I am saying is I love Orcaslicer and really dont want to have to stop using it just because of this.

I have dealt with it for three month's now, and finally decided let me try to put in a bug as the discord no one is helpful.

But yeah if I can light a fire under someone who can assist I would because otherwise the slicer is near useless to me at this point.

xorinzor, 2023-12-20:
I am just telling you that an Issue Tracker is not a community section or forum. You report your bug, add any relevant technical information and diagnosing you did, and leave it at that.

We now already have 6 comments (including this one) that's clouding up this specific issue.
Like I said, it is free software, people are working on this in their spare time. It sucks that you've had this issue for 3 months, but you only raised this issue 4 days ago.

I would suggest you use Bambu Slicer in the meanwhile and just keep an eye on the releases to see if any future updates fix it; or downgrade to a previous version that didn't have this issue (I used 1.6.x before and didn't have the crashing). The issue will get adressed eventually.

silence48, 2023-12-31:
Likely related to issue [2921](https://github.com/SoftFever/OrcaSlicer/issues/2921) also.  I've left some comments there regarding it.

closed 2024-04-07 by github-actions[bot]