OrcaSlicer/OrcaSlicer #3158: Doesn't slice anymore. A fatal error occurred: "Access violation"

FillPavS, 2023-12-17:
### OrcaSlicer Version

from Actions (Full re-write of spiral vase (#3091))

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

Open the slicer, add any primitive or model, slice it.

### Actual results

A fatal error occurred: "Access violation". There are no errors on the earlier version dated 09 Dec 23 (Port Emboss & SVG gizmo from PrusaSlicer (#2819))

[image]

### Expected results

Obviously there should be no error.

### Project file & Debug log uploads

[debug_Sun_Dec_17_11_07_18_1396.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/13695191/debug_Sun_Dec_17_11_07_18_1396.log.0.zip)
[Cute Baby Dragon Fauna.zip](https://github.com/SoftFever/OrcaSlicer/files/13695192/Cute.Baby.Dragon.Fauna.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2023-12-17:
Thank you.
Fixed in this build: https://github.com/SoftFever/OrcaSlicer/actions/runs/7238476339

closed 2023-12-17 by FillPavS

tlhintoq, 2023-12-22:
So if this was fixed in the build of 23-12-17...
Shouldn't it be part of 23-12-21?  Because I'm getting the same fatal exception in 23-12-21
[image]

tlhintoq, 2023-12-22:
@SoftFever @Noisyfox 
I'm still getting this `Access Violation` bug with newer builds.
For example:  Nightly build 2312122 is several days newer. Shouldn't this fix be included to newer builds?
https://github.com/SoftFever/OrcaSlicer/actions/runs/7294849330
[image]

tlhintoq, 2023-12-22:
@Noisyfox @SoftFever 
Might be worth noting that the nightly build linked above has the error yet this PR release slices fine.
https://github.com/SoftFever/OrcaSlicer/actions/runs/7300472441

SoftFever, 2023-12-23:
> @Noisyfox @SoftFever Might be worth noting that the nightly build linked above has the error yet this PR release slices fine. https://github.com/SoftFever/OrcaSlicer/actions/runs/7300472441

Thank you.
Can you shared your project file?

tlhintoq, 2023-12-23:
[Square_Nathans.zip](https://github.com/SoftFever/OrcaSlicer/files/13757896/Square_Nathans.zip)
@SoftFever There ya go.  But it was happening with anything I dropped on a fresh build plate.

RoyMarok, 2024-06-17:
Hi. Got this bug on 2.0.0. Perfect working since 12.05.2024. Start crushing on slicing step 16.06.2024. In 1-st half of day of 17.06.2024 work as normal. I think this issue local time depended. 
AMD Ryzen 7 7700X, RAM 64, Win10, AMD Radeon 6650XT, Creality Ender 3v3KE