OrcaSlicer/OrcaSlicer #3160: OrcaSlicer doesn't start on MX Linux

Newklolz, 2023-12-17:
### OrcaSlicer Version

1.8.1 and lower

### OS version

MX-23.1_ahs_x64

### Additional system information

DELL Precision 5530 with NVIDIA Quadro P1000

[CODE]Graphics:
Device-1: Intel CoffeeLake-H GT2 [UHD Graphics 630] vendor: Dell driver: i915 v: kernel
arch: Gen-9.5 process: Intel 14nm built: 2016-20 ports: active: DP-4 off: eDP-1 empty: DP-1,
DP-2, DP-3, DP-5, DP-6 bus-ID: 00:02.0 chip-ID: 8086:3e9b class-ID: 0300
Device-2: NVIDIA GP107GLM [Quadro P1000 Mobile] vendor: Dell driver: nvidia v: 525.147.05
non-free: 530.xx+ status: current (as of 2023-03) arch: Pascal code: GP10x process: TSMC 16nm
built: 2016-21 pcie: gen: 1 speed: 2.5 GT/s lanes: 16 link-max: gen: 3 speed: 8 GT/s
bus-ID: 01:00.0 chip-ID: 10de:1cbb class-ID: 0302
Device-3: Realtek Integrated_Webcam_HD type: USB driver: uvcvideo bus-ID: 1-12:4
chip-ID: 0bda:5657 class-ID: 0e02 serial:
Device-4: Lenovo 500 RGB Camera type: USB driver: uvcvideo bus-ID: 3-2.3.3.3.1:12
chip-ID: 17ef:482f class-ID: 0e02 serial:
Display: x11 server: X.Org v: 1.21.1.7 compositor: xfwm v: 4.18.0 driver: X:
loaded: modesetting,nvidia unloaded: fbdev,nouveau,vesa alternate: nv dri: iris gpu: i915
display-ID: :0.0 screens: 1
Screen-1: 0 s-res: 5120x1440 s-dpi: 96 s-size: 1355x381mm (53.35x15.00")
s-diag: 1408mm (55.42")
Monitor-1: DP-4 mapped: DP-1-1 pos: primary model: Samsung C49RG9x serial: built: 2018
res: 5120x1440 hz: 60 dpi: 109 gamma: 1.2 size: 1190x340mm (46.85x13.39") diag: 1239mm (48.8")
modes: max: 3840x1080 min: 720x400
Monitor-2: eDP-1 note: disabled model: Sharp 0x149a built: 2018 res: 5120x1440 dpi: 142
gamma: 1.2 size: 344x194mm (13.54x7.64") diag: 395mm (15.5") ratio: 16:9 modes: 1920x1080
API: OpenGL v: 4.6 Mesa 23.1.2-1~mx23ahs renderer: Mesa Intel UHD Graphics 630 (CFL GT2)
direct-render: Yes[/CODE]

### Printer

Prusa

### How to reproduce

1. Install fresh MX Linux (23.1_ahs_x64) on bare metal or virtual machine.2. 
2. Download newest AppImage.
3. Make it executable.
4. Run it in the terminal.

### Actual results

Nothing happens after the SSL cert screen.

$ ./OrcaSlicer_Linux_V1.8.1.AppImage 
[2023-12-17 11:29:10.291478] [0x00007f9d8c7d4ec0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
loading network module at /home/.../.config/OrcaSlicer/plugins/libbambu_networking.so
error, dlerror is /home/.../.config/OrcaSlicer/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(orca-slicer:3354): Gtk-CRITICAL **: 11:29:13.420: gtk_window_resize: assertion 'width > 0' failed

(orca-slicer:3354): GLib-GObject-WARNING **: 11:29:13.440: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:3354): Gtk-CRITICAL **: 11:29:13.441: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:3354): GLib-GObject-WARNING **: 11:29:13.442: invalid cast from 'wxPizza' to 'GtkCellLayout'

[... 89 lines cut ...]

OrcaSlicer should start.

### Project file & Debug log uploads

(gdb) bt
#0  0x0000555556f0b751 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()
#1  0x0000555556e9fe1c in Slic3r::GUI::StatusBasePanel::create_monitoring_page() ()
#2  0x0000555556ea5a7b in Slic3r::GUI::StatusBasePanel::StatusBasePanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) ()
#3  0x0000555556eadcca in Slic3r::GUI::StatusPanel::StatusPanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) ()
#4  0x0000555556a7506e in Slic3r::GUI::MonitorPanel::init_tabpanel() ()
#5  0x0000555556a757f3 in Slic3r::GUI::MonitorPanel::MonitorPanel(wxWindow*, int, wxPoint const&, wxSize const&, long) ()
#6  0x0000555556897e55 in Slic3r::GUI::MainFrame::init_tabpanel() ()
#7  0x00005555568994d6 in Slic3r::GUI::MainFrame::MainFrame() ()
#8  0x0000555556856bcc in Slic3r::GUI::GUI_App::on_init_inner() ()
#9  0x0000555556857dda in Slic3r::GUI::GUI_App::OnInit() ()
#10 0x0000555559133ac7 in wxEntry(int&, wchar_t**) ()
#11 0x000055555681c84b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#12 0x0000555555eaa877 in Slic3r::CLI::run(int, char**) ()
#13 0x0000555555dd3af9 in main ()

### Checklist of files to include

- [X] Log file
- [X] Project file

closed 2023-12-19 by Newklolz

alive-corpse, 2024-05-05:
+1 

I'm also using mx linux 23 and installing  gstreamer1.0-x solved the same problem for me.

Thank you very much!

Constant-Leon, 2024-09-20:
+1 This exact issue just happened, solved using the guidance above.

blaa, 2025-02-25:
This is relevant on Debian Trixie with OrcaSlicer V2.2.0. Installing gstreamer package helps.  The error is really cryptic and hard to pinpoint - I was assuming it's caused by XMonad, but it wasn't.

Tasmedic, 2025-04-06:
Just a thanks for this help. The files ought to be there in the appimage I guess.