OrcaSlicer/OrcaSlicer #3165: Custom color - HEX value

Eldenroot, 2023-12-17:
**Is your feature request related to a problem? Please describe.**
Some filament prducers use HEX colors for their filaments (color), anyway I cannot add it into the database etc. I know, we can convert it but why to make it difficult?

**Which printers will be beneficial to this feature**
All, slicer based.

**Describe the solution you'd like**
Add a way to choose my own color with HEX values (
[image]
)

**Describe alternatives you've considered**
Add a new row - add hex.

**Additional context**

joesanford, 2023-12-17:
I'm pretty sure that color pickers are OS-specific and not part of this codebase.

Eldenroot, 2023-12-17:
Hmm, OK, so use a 3rd library or close the issue. I am closing this :(

joesanford, 2023-12-17:
[image]
Or switch operating systems - it's possible on macOS.

Eldenroot, 2023-12-17:
Nope, unfortunately SW for my work is not available for MacOS :)

Anyway thank you :)

Eldenroot, 2023-12-17:
Well, actually I am reopening this one - because Windows 11 contains a new color picker UI; however Bambu Lab Studio/Orca slicer use the old picker in the codebase. It must be activated 

https://learn.microsoft.com/en-us/uwp/api/windows.ui.xaml.controls.colorpicker?view=winrt-20348

Then it looks like this one:
[image]

closed 2024-01-25 by Eldenroot