OrcaSlicer/OrcaSlicer #3168: Bambulab A1 (not Mini) Support/Printerprofile

CoReYeDe, 2023-12-17:
Actually i cant find a possibility to import the BBL A1 Printer-Profile exported from BambuStudio (Filetype: bbscfg)
Also no predefined Profile for the new A1-Printer (not Mini)

Is a Update planned with this Profile or how can i get Support for this Printer in OrcaSlicer <3

Thanks

Eldenroot, 2023-12-17:
Orca needs to merge latest version 1.8.2 of Bambu lab studio... few commits, so I think it will happen soon :) I hope so, because A1 is on the way to my home :D

ForsakenHarmony, 2023-12-21:
Just got mine and was hoping OrcaSlicer already supported it.
Well, soon :tm:

Eldenroot, 2023-12-21:
Already supported, download nightly build

ForsakenHarmony, 2023-12-21:
@Eldenroot ah yes just found it, thanks

CallmeBigfat, 2023-12-21:
@Eldenroot hey where would one find the nightly builds? also would like to use my standard A1.

closed 2024-03-29 by github-actions[bot]