OrcaSlicer/OrcaSlicer #3169: Split to parts misses one of the parts

ctadlock, 2023-12-17:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11

### Additional system information

AMD Ryzen, 16GB

### Printer

Bambu X1C and P1S

### How to reproduce

1. Create a multi body model; in my example I have a base model "spacer" with embossed text that has been filled in to with another body "text". 
2. In the slicer, "split to parts" or "split to objects"
3. Look at the parts the slicer detected

[image]

### Actual results

The slicer does not detect one of the bodies. This is not an issue with the actual model. Opening the exported stp file directly shows the appropiate solid bodies. The slicer seems to not detect either the first or last one. I have proven this by creating a test where I add another solid body "test" with the same embossed design; in this case it got the text bodies fine but not the new test body.

[image]

[image]

### Expected results

The slicer should detect all solid bodies. 

This is a relatively new bug in the last couple releases. I have printed this same part many times before without issue until recently. 

### Project file & Debug log uploads

[roller_spacer_10_test.zip](https://github.com/SoftFever/OrcaSlicer/files/13697837/roller_spacer_10_test.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ctadlock, 2023-12-17:
Here is a clue to where the bug is. If I dont make the body fill in the emboss fully, but leave it recessed then the slicer properly detects it. 

[image]

ctadlock, 2023-12-17:
Same issue with BambuStudio 1.8.2

[image]

ctadlock, 2023-12-18:
If I repair the model in the slicer, it removes the "part".

[image]

EricCraigen, 2023-12-18:
I am having this issue with multi-body models too.

avxmw, 2024-11-22:
Same problem here. I'm having tool holders which are supposed to be labeled on both sides, however on body is always missing.

When I label them 6/6 (left/right side) i only get 6 on one side, if I label them 23/23 I get 23/2 which leads me to assume there is an off-by-one error in processing bodies not actually an identification problem.

PR #8084 (2025-01-22): FIX: step model split COMPOUND and COMPOUNSOLID type
Fix #3169

cherry picked from bambulab/BambuStudio@90f325792dc4910b80d69c9d6a5677f24b8fa259 
Thanks BambuLab!
closed 2025-01-22 by Noisyfox