OrcaSlicer/OrcaSlicer #3170: Do not flush support interface material into support/infill/object

Evil-Crab, 2023-12-17:
**Is your feature request related to a problem? Please describe.**
Currently, during the multi-material printing, support interface material is flushed into support/infill/object if the corresponding option is turned on. Since support interface materials are designed to not fuse with the primary material this leads to weakened support, that can collapse on its own, weakened infill, or fragile object used for flushing.

**Which printers will be beneficial to this feature**
All printers that can utilize multiple materials.

**Describe the solution you'd like**
Material set as "support interface materials" should be excluded from the list of materials flushable into support/infill/object. This can be a configurable parameter, although I don't see any reason to disable that.

**Describe alternatives you've considered**
Flushability into support/infill/object can also be a part of the filament config, but IMHO it's not the best solution since some regular materials can be used as support interfaces (e.g. PLA for PETG and vice versa).

**Additional context**
I experienced this issue personally while printing the PETG part with PLA as a support interface and the "flush into support" option enabled. Several supports collapsed because of the weak PLA layers in the middle of them.

[image]

Evil-Crab, 2023-12-20:
Ok, it seems that filament marked as support material in the filament config is already excluded from flushing into things, but I think that the filament set as support interface should also be excluded regardless of whether it's marked as support in the filament config. 

https://github.com/SoftFever/OrcaSlicer/blob/6bb0d5ec30559f9d54556da1e55d484363ed090b/src/libslic3r/GCode/ToolOrdering.cpp#L1003

This will help to avoid the cases where the regular filament is used as the support for the different material type, (e.g. PLA for PETG).

TheAtomicSoul, 2024-01-09:
I second this, but also have a question: What setting does labeling something as support change?

closed 2024-07-16 by github-actions[bot]

Alkerion, 2024-08-29:
+1
And no flush at all in purge tower too....
[image]