OrcaSlicer/OrcaSlicer #3171: Bambu lab X1C. When sending a print job, progress gets stuck at 10% on some models

FillPavS, 2023-12-18:
### OrcaSlicer Version

1.9.0-dev (Fixed a regression that non spiral vase mode was broken. #3091)

### OS version

Windows 10

### Printer

Bambu lab X1C

### How to reproduce

Open the slicer, load the attached model or project, slice, send for printing

### Actual results

Progress gets stuck at 10%. You can wait forever, but nothing happens. If you do the same in the earlier version (dated 09 Dec 23 (Port Emboss & SVG gizmo from PrusaSlicer (https://github.com/SoftFever/OrcaSlicer/pull/2819)), then everything works as it should.
[image]

### Expected results

This should work

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13699679/log.zip)
[project.zip](https://github.com/SoftFever/OrcaSlicer/files/13699683/project.zip)
[model.zip](https://github.com/SoftFever/OrcaSlicer/files/13699687/model.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

FillPavS, 2023-12-22:
Initially, I made a mistake in determining the version where everything worked and was able to determine the version where it did not work.
Work in - Fixed a bug that caused curled edge detection not to work as expected… https://github.com/SoftFever/OrcaSlicer/actions/runs/7150314111
Doesn`t work in - Port Emboss & SVG gizmo from PrusaSlicer (#2819) and later
Both versions from December 9

Noisyfox, 2023-12-22:
Will take a look.

PR #3237 (2023-12-23): Trigger task progress update on paint event as well
This fixes #3171
closed 2023-12-23 by SoftFever