OrcaSlicer/OrcaSlicer #3173: When previewing slices, never reset the view when changing a setting

katrinleinweber, 2023-12-18:
**Is your feature request related to a problem? Please describe.**

The human visual cortex suffers from "[change blindness](https://en.wikipedia.org/wiki/Change_blindness#Research_in_the_1990s_and_2000s)" when between 2 different images, a 3rd grey/blank one is inserted. It is therefore difficult to impossible to see (and thus quickly understand) the impact of some change in the slicing settings on the model to-be-printed, because the (pre)view is reset upon each change.

**Which printers will be beneficial to this feature** All

**Describe the solution you'd like**

Once the slicing preview is enabled, any change to settings should only _update_ preview _in-place_, ideally with chaching of the old data, so that toggling the same slicing setting back-and-forth causes immediate visual feedback.

**Describe alternatives you've considered**
Does any slicer do something like that? I've checked Slic3r and Cura, and both suffer from the same problem.

**Additional context**
Or am I missing something (a setting, print prep strategy, etc?) that's obvious to more advanced OrcaSlicer users, and would circumvent my problem here?

igiannakas, 2023-12-19:
The issue why this works the way it does, is because when you change a setting the gcode that you are previewing is invalid. So it’s cleared and regenerated. One way to circumvent it would be to do all your changes on the model screen and then click slice once.

katrinleinweber, 2023-12-19:
> So it’s cleared and regenerated

Yes, but exactly that "clear"ing is making it hard to see the difference from old/invalid to new/regenerated gcode  (or rather: its image preview/representation).

cdvankammen, 2024-01-13:
This is an interesting problem I had not thought about. It is true that if there was an option for it to generate in the background if someone were to so desire it could be an experimental SLICER SETTINGS option? (ALSO auto regenerating should be an option as in prusa slicer)

closed 2024-04-20 by github-actions[bot]

cdvankammen, 2024-04-20:
This should be a feature

katrinleinweber, 2024-04-20:
Hi @igiannakas 👋 Could you please re-open this, as there are a few interested people?