OrcaSlicer/OrcaSlicer #3190: please add release binaries for Linux ARM/ARM64

wildblue, 2023-12-19:
**Which printers will be beneficial to this feature**
a supported printers

**Describe the solution you'd like**
The availability to download a binary release for Linux ARM/ARM64 which can be executed on a Raspberry Pi

**Describe alternatives you've considered**
Buying a WibdowsPC, but NAH!

**Additional context**
Please Support Open Operating Systems based on ARM/ARM64 also on Linux (as there are binaries for MacOS available right now).

Botspot, 2024-02-16:
Interested in adding OrcaSlicer to [our ARM Linux app store.](https://github.com/Botspot/pi-apps)

Gismo6303, 2024-03-05:
Hi, I'm for an ARM version too, I want to test it on my raspi 5

Botspot, 2024-03-05:
@SoftFever, if you do not have access to an ARM system, I can offer access to my RPi5 using Rustdesk. Please respond if that would be helpful, or if it would be helpful for me to share some good docs for setting up an ARM building environment on Github Actions.

Gismo6303, 2024-03-06:
Thx for the discover off pi-apps! I can't offert access on my raspi, but I can compile and test on it if you want, I'm not a Linux boss, but I try to do my best

Gismo6303, 2024-03-31:
Hi here radek6ak has an answer for us, we can compile a working appimage on raspi see here
https://github.com/SoftFever/OrcaSlicer/discussions/1425#discussioncomment-8686064

If someone tells me which files I've to upload I can do it if you want the appimage

ClaytonRehder, 2024-08-04:
Any updates about the Linux ARM version? Cura, Bambu Studio and Prussa Slicer already have it on flatpak

Botspot, 2024-11-03:
@SoftFever I have offered you remote desktop access to my hardware. Does this interest you or not?

claywd, 2024-12-26:
I need this. I have hardware and experience. What's needed here?

cochcoder, 2025-01-10:
It seems like we're going to have to wait for github to make it publicly accessible to all accounts. The current ETA seems to be sometime Q1 2025. You can find the roadmap issue for it [here](https://github.com/github/roadmap/issues/970).

cochcoder, 2025-02-28:
I have successfully made a arm64 flatpak build of Orca Slicer in PR #8033. Any tests are greatly appreciated as the only arm computer I own is a Raspberry pi 3 b+ and I'm not sure how many bugs are from it being slow and having only 1GB of ram. The Flatpak file for arm can be found [here](https://github.com/SoftFever/OrcaSlicer/actions/runs/13580568050?pr=8033).

Botspot, 2025-03-01:
> I have successfully made a arm64 flatpak build of Orca Slicer in PR [#8033](https://github.com/SoftFever/OrcaSlicer/pull/8033). Any tests are greatly appreciated as the only arm computer I own is a [...]

After downloading [OrcaSlicer-Linux-flatpak_PR-8033-aarch64.flatpak](https://github.com/SoftFever/OrcaSlicer/actions/runs/13580568050/artifacts/2668196882) and extracting, I did `sudo flatpak install /home/pi/Downloads/OrcaSlicer-Linux-flatpak_PR-8033.flatpak`, which seems to work on my Pi5 8GB.
It is running under x11, which makes it slow and breaks drag-and-drop on the latest Pi OS. Not sure if Wayland is an option, but if it is it would make a huge difference. Hope this helos!

[image]

cochcoder, 2025-03-02:
> After downloading [OrcaSlicer-Linux-flatpak_PR-8033-aarch64.flatpak](https://github.com/SoftFever/OrcaSlicer/actions/runs/13580568050/artifacts/2668196882) and extracting, I did `sudo flatpak [...]

Thank you for your reply! Are you able to successfully upload a print to your printer? As that seems to be the only bug that needs fixing. 

I'll definitely look into adding native wayland support in a future PR, however this is most likely not in my skill level.

Botspot, 2025-03-02:
> Thank you for your reply! Are you able to successfully upload a print to your printer? As that seems to be the only bug that needs fixing.
> 
> I'll definitely look into adding native wayland support in a future PR, however this is most likely not in my skill level.

I use Ender 3 and save to SD card, so I cannot check if uploading works.

cochcoder, 2025-03-09:
Have marked my PR #8033 as ready to merge, hopefully everything goes through without issue and we get a arm flatpak!