OrcaSlicer/OrcaSlicer #3191: The portable version of OrcaSlicer isn't portable.

mogrol, 2023-12-19:
### OrcaSlicer Version

1.8.1

### OS version

Windows 10

### Printer

Creality Ender 3 v3 se

### How to reproduce

- Download the portable version of (Orca Slicer [OrcaSlicer_Windows_V1.8.1_portable.zip](https://github.com/SoftFever/OrcaSlicer/releases/download/v1.8.1/OrcaSlicer_Windows_V1.8.1_portable.zip) was used in this case)
- Extract Orca Slicer to any folder on a computer A's hard drive.
- Open Orca Slicer
- Set up the printer profile
- Close Orca Slicer
- Copy the folder to computer B
- Open Orca Slicer on computer B

### Actual results

None of the previously created configurations are stored in the folder the program was run in.

### Expected results

All created profiles, configuration and so on should be stored in the folder the program was run in.

A program which advertises itself as portable should store all configuration, settings, profile and so on in the folder/directory it's run in.

### Project file & Debug log uploads

No project file is needed to reproduce this issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ilko-k, 2023-12-19:
Here is a temporary solution until it is fixed. I wrote a simple batch script to backup/restore the Orca profile. The batch file can be placed anywhere before the backup, but the backup folder must be in the same place as the script afterwards. Nothing will be done if there is nothing to do.

[image]

[Orca_profile.zip](https://github.com/SoftFever/OrcaSlicer/files/13717909/Orca_profile.zip)

Noisyfox, 2023-12-20:
One possible way to implement a real portable mode is: When Orca is startup, it checks if there is a `.portable` file next to the exe. If yes, then it puts the data folder under current dir, otherwise uses appdata folder as usual.
There will be some other issues need to be solved, such as path to customized bed texture/model that stored as abs path in profiles etc.

@SoftFever thoughts?

SoftFever, 2023-12-20:
We can use the '—datadir’ parameter when start orca-slicer.exe. 
that should work for you?

tome9111991, 2023-12-20:
I like the solution from noisyfox.

I think if you use the --datadir you need always a absolute path

javales74, 2023-12-20:
> We can use the '—datadir’ parameter when start orca-slicer.exe. that should work for you?

In the lastest nightlies this option doesn't work.
#3132

abstractFlo, 2023-12-24:
> > We can use the '—datadir’ parameter when start orca-slicer.exe. that should work for you?
> 
> In the lastest nightlies this option doesn't work. #3132

+1 for this, this is an essential feature for me, i've been using different instances for each of my printers. But currently i need the dev branch for my A1 Printer. Hopefully it can be fixed 🙈

SoftFever, 2023-12-24:
I will take a look, thanks for the feedback

closed 2023-12-27 by SoftFever

mogrol, 2023-12-28:
Is the proposed solution to ask the user to add `--datadir` with a fixed path when launching Orca Slicer? If so, it'll still not be a portable application.

AJolly, 2024-03-08:
I'd prefer there be an option to check for the .portable file, not just pass a command line parameter to the exe.  That way even if you run the exe directly, it'll still be in portable mode.

MelGrubb, 2024-11-20:
I have two different laptops that regularly come into play at home, but only a single printer. When I make calibration changes or add a new filament type from one computer, I really need those changes to carry over to the other one. To add difficulty, the path to my portable applications is not the same across both computers. One has a single, 2TB SSD, and everything is "C" (because Windows). The other has two 1TB SSDs, and my portable stuff is all on the second one, "D". An absolute path really doesn't work well in this case, so I'd need to create separate shortcuts for each computer to make it work. That's not hard, but it's shifting some of the work of making the app portable onto each and every user of the system. Each and every portable user ends up doing the same search and ending up on this same page reading this same conversation.

Allowing the app to realize "Oh, look, a 'Settings' folder... I think I'll use that" makes it automatic for every user. It's a classic case of fixing it once or explaining it forever. I think the --datadir parameter definitely needs to stay in order to account for other edge cases, but by default, if there's a folder called "Settings" or "Portable", or whatever you settle on, then just use that. When you create the zip file or portable installer, it just needs to create that folder, and maybe copy some initial contents into it. I'm a developer myself, so I realize that nothing's ever really that simple, but we do the hard part so that the users can _not_ appreciate the work that was done. That's our destiny.

commit bab1a795 (2023-12-27): Fixed a regression bug that --datadir was broken
fixed #3191
commit a9c0e3a1 (2023-12-30): Fixed a regression bug that --datadir was broken
fixed #3191