OrcaSlicer/OrcaSlicer #3201: Mouse ears not attached to part

quantumfrost, 2023-12-20:
### OrcaSlicer Version

1.8.1

### OS version

Windows

### Printer

All

### How to reproduce

Select mouse ears as bed adhesion type
Slice a model that has a thin wedge that tapers to a point

### Actual results

[image]

### Expected results

Mouse ear attached to the part

### Project file & Debug log uploads

[thin wedge.zip](https://github.com/SoftFever/OrcaSlicer/files/13723789/thin.wedge.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-12-20:
They are - but they also are not. This is happening because your part is not printed completely hence the mouse ears attempt to attach to an area that is ignored when slicing. Select Arachne to get a chance at printing the complete part. Or alternatively attach a mouse ear manually - add a small cylinder 1 layer tall and place it by hand. This is unlikely to be fixed as the code ads the mouse ears on the assumption that the whole part will be printed.

quantumfrost, 2023-12-21:
> They are - but they also are not. This is happening because your part is not printed completely hence the mouse ears attempt to attach to an area that is ignored when slicing. Select Arachne to get [...]

I understand why they are not being attached. You can see the shadow of the actual geometry in the image (this is with arachne enabled). My feature request is that the code should be aware of this and position the mouse ears accordingly. Super thin areas are exactly what needs mouse ears the most.

closed 2024-03-29 by github-actions[bot]