OrcaSlicer/OrcaSlicer #3211: There is a way to completely eliminate the visible seam.

vgdh, 2023-12-20:
**Is your feature request related to a problem? Please describe.**
# Hide the Seam Completely
## Explanation
Every closed loop will eventually have a visible seam. However, if there is no specific point where the line ends, there is no seam, right?
1. Begin the line with a height of 0 and gradually increase the height of the nozzle and the line over some length until it reaches the full height of the line.
2. Move through the loop until reaching the end of the line, but continue extruding the filament due to the initial zero line height. Systematically reduce the line's height while maintaining the same nozzle height until reaching the end of the starting slope.
3. Move on to the next perimeter or infill.

That's it. The algorithm is remarkably simple, though the result is astonishingly good. 
It is not an easy task to post-process already existing GCODE files, but if it works at the slicer level, it will work very well.

## The python code
The repository includes an [example](https://github.com/vgdh/seam-hiding-whitepaper/blob/main/main.py) demonstrating a basic implementation of the algorithm for printing a ring. This implementation allows specifying parameters such as the number of lines in the ring, layer height, line width, speed, etc.

Also there is an [example](https://github.com/vgdh/seam-hiding-whitepaper/blob/main/nozzle%200.4%20line%200.3x0.8.gcode) of gcode file for 0.4 nozzle line 0.3x0.6 layer height 0.3

## Photos of the Seam Details on the Printed Example.
The first layer (the top on the first photo) was printed without the algorithm to get better adhesion.
[image]
[image]
[image]

In the photo below you can see there is no gap between the lines.
[image]
[image]
[image]
[image]
[image]

**Which printers will be beneficial to this feature**
all the comunit

vgdh, 2023-12-20:
printed on the cheapest kingroon kp3s

Eldenroot, 2023-12-20:
Nice idea :)

TheBest6337, 2023-12-20:
This could be game changing! Idk how well it could work on smaller things and not smooth shapes but it would be nice to have an implementation to test.

Noisyfox, 2023-12-24:
One possible issue I could think of is the nozzle clearance, since you'll have to lower the z every time you start printing an outer wall.

PrimeGoat, 2023-12-25:
How does this differ from the spiral vase?

vgdh, 2023-12-26:
I have written some post-processing [script](https://github.com/vgdh/seam-slope-postprocessor/blob/main/postprocessor_seam_slope.py). Feel free to test, modify, or contribute.

DVSVIDEO, 2023-12-26:
> Hogyan különbözik ez a spirális vázától?

> How does this differ from the spiral vase?

In that the spiral vase mode can only print a single object that returns to itself within the slice. So an object branching into two separate branches cannot be created with a single line.
This method works for any number of objects.

SoftFever, 2023-12-29:
This is pure genius!

Eldenroot, 2023-12-29:
This could be a game changer

Noisyfox, 2023-12-29:
https://github.com/Noisyfox/seam-slope-postprocessor/tree/dev/better-orca-support

I've made a version that should be (mostly) usable with Orca as a post-processor script, with the following requirements:

- Arc fitting: off
- Wipe distance: 0
- Z hop when retracting: 0
- Disable set remaining time
- Seam gap: 0

This script is still very prototype, use at your own risk!

Eldenroot, 2023-12-29:
Omg cool! Good job!

vgdh, 2024-01-01:
Now the script processes only the outer perimeters, printing the inner ones as usual.

vgdh, 2024-01-01:
bender likes it very much 😁
[image]
[image]

Eldenroot, 2024-01-01:
Wau!

Noisyfox, 2024-01-02:
Looks promising, will give it another shot tonight.

Eldenroot, 2024-01-21:
Did you try and did you get better results?

Noisyfox, 2024-01-22:
Yeah and I still got that over extrusion on overhangs... Maybe something wrong with my PA.

Eldenroot, 2024-01-22:
Ouch, not good :( because this  would be a huge step forward in FDM printing world. 

@vgdh - do you have any idea?

Noisyfox, 2024-01-26:
I've successfully integrated this into the slicer itself:

[image]

Eldenroot, 2024-01-26:
Amazing!

superfloh247, 2024-01-26:
please push a PR - I want to test this!

Noisyfox, 2024-01-26:
Ok the PR is raised.

superfloh247, 2024-01-26:
git fetch origin pull/3839/head:pr3839
git checkout pr3839
./build_release_macos.sh -t 14.2

correct?

is this new feature activated by default or how do I switch it on?

Eldenroot, 2024-01-26:
Just download the build from PR

vgdh, 2024-01-27:
> git fetch origin pull/3839/head:pr3839 git checkout pr3839 ./build_release_macos.sh -t 14.2
> 
> correct?
> 
> is this new feature activated by default or how do I switch it on?

[win pr artifact](https://github.com/SoftFever/OrcaSlicer/actions/runs/7675961229/artifacts/1199247257)

vgdh, 2024-01-31:
This feature in now implementing by OrcaSlicer: https://github.com/SoftFever/OrcaSlicer/pull/3839
Feel free to participate.

vgdh, 2024-02-27:
> Thank you!

You are welcome! ❤️

closed 2024-03-02 by SoftFever