OrcaSlicer/OrcaSlicer #3214: 2D GUI buffers not being drawn after DPMS off

evranch, 2023-12-21:
### OrcaSlicer Version

1.8.1

### OS version

Manjaro Plasma

### Additional system information

Intel i3-12100F
32GB RAM
Radeon RX 6700XT

### Printer

TronXY X5SA Pro

### How to reproduce

Add a STL to plate
Suspend and resume PC or allow monitor to enter DPMS off state while idle
3D display updates and can be manipulated but 2D regions are glitchy, sometimes functional but not drawn properly.
Often content from behind the window will be mirrored into these areas, and when moving the window that mirrored content will be drawn instead of the interface.

### Actual results

[image]
[image]

### Expected results

App should return to normal functionality. This PC does not have any other graphical issues, with good gaming performance and no issues with any other applications returning from suspend.

### Project file & Debug log uploads

[12v_pump_sleeve.zip](https://github.com/SoftFever/OrcaSlicer/files/13733928/12v_pump_sleeve.zip)
[debug_Wed_Dec_20_16_17_35_27619.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/13733938/debug_Wed_Dec_20_16_17_35_27619.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-12-21:
Have you tried the latest master build?

evranch, 2023-12-21:
I have not tried to compile from source, I've only ran the AppImage releases. This issue has persisted for me in every release since I started using OrcaSlicer at v1.8.0-beta. 

I'll try to build it from master and let you know what happens.

Noisyfox, 2023-12-21:
You don't need to compile it yourself. You can get them here https://github.com/SoftFever/OrcaSlicer/actions

evranch, 2023-12-21:
Excellent, as the build script assumes a Debian-based system and I was just modifying it to get it to run. Much easier.

It seemed to work but now I'm having trouble intentionally replicating the bug on 1.8.1 by suspending the PC. I'll leave the master build running in the background and see if it occurs, as on previous versions it's always been glitched when I walked away and came back to a sleeping display. Thanks

evranch, 2023-12-23:
I've run it for several days, a couple prints and multiple suspends, without having to restart the app. Must have been fixed either here or upstream... closing

evranch, 2023-12-30:
Looks like I closed this too early as the bug recurred today. Interestingly I was running PrusaSlicer and OrcaSlicer at the same time to compare slicing performance, and both were affected in the same way, meaning this bug likely comes from somewhere upstream.

[Archive.zip](https://github.com/SoftFever/OrcaSlicer/files/13799036/Archive.zip)

closed 2024-04-06 by github-actions[bot]