OrcaSlicer/OrcaSlicer #3217: inner-outer-inner/infill missing

superfloh247, 2023-12-21:
### OrcaSlicer Version

401ac1a

### OS version

MacOS 14.2.1

### Printer

Bambu P1P

### How to reproduce

click on dropdown

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

n/a 

### Checklist of files to include

- [X] Log file
- [X] Project file

Eldenroot, 2023-12-21:
This was changed in https://github.com/SoftFever/OrcaSlicer/pull/2962

Maybe some regression?

@igiannakas

igiannakas, 2023-12-21:
No, this is deliberate. If you want infill first tick the box right under the wall order. Ported from BBL slicer a while ago.

closed 2023-12-21 by superfloh247