OrcaSlicer/OrcaSlicer #3219: Strange Diagonal Line at 50% of the print

MrBlonde1989, 2023-12-21:
### OrcaSlicer Version

1.8.1

### OS version

Windows 11 23H2

### Additional system information

CPU : AMD Ryzen 7 5800X 8-Core
4x 8GB RAM DDR4 3600Mhz
Nvidia 3060 Ti

### Printer

Sovol SV06 Klipperized and enclosed

### How to reproduce

Hello everyone,

The issue occurred during the calibration of the PETG filament flow.

It seems to occur only on one layer, and I don't understand why this behavior only manifests itself halfway through the print, around 50%.

The object is still usable after printing, but at the end of the print, I noticed that there is a problem with the underlying layers. It's not catastrophic, but I wanted to understand if I'm doing something wrong in orcaslicer or if it could be a bug. I have set up the Z-hop, and it executes it in all the other steps, but for some reason, only for this particular movement, it moves flush on the piece, back and forth, creating this line. This line not only scratches the piece but also dirties the nozzle. I don't understand why it is executed. I have attached videos to help you understand better:

I tried the same model on Prusa Slicer and the issue does not occur.

Has anyone else encountered this problem?

What do you think is the best way to solve it?

### Actual results

Here is on slicer:
[https://youtu.be/fH2t155aouQ](https://youtu.be/fH2t155aouQ)

Here is the print movement:
[https://youtu.be/aUh30w97gwE](https://youtu.be/aUh30w97gwE)

there is a scretch on the fourth layer on surface of the print 
[image]

### Expected results

the fourth layer without scretch

### Project file & Debug log uploads

[Flowrate Test - Pass1.zip](https://github.com/SoftFever/OrcaSlicer/files/13740564/Flowrate.Test.-.Pass1.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13740565/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2023-12-21:
It’s because z hop is not enabled.

MrBlonde1989, 2023-12-22:
On my profile there are:
Printer Settings:
[image]
Filament Settings override:
[image]
at this point, where i can enabled It?

igiannakas, 2023-12-22:
It’s disabled because your only lift z below is 0. Set it to your max print height minus your z hop height

MrBlonde1989, 2023-12-22:
In the first place, I thank you for the support you are giving me.
I'm not sure if I understood correctly. The maximum Z position for my printer is 260 mm, with a 2mm Z-hop height when retracting. I have edited the Filament settings overrides as follows:
[image]
But it seems that nothing has changed:
[image]
It seems that the other movements experience Z-hop, except for the diagonal line.

igiannakas, 2023-12-22:
This is what I mean here::

set the only lift z below to 258

[image]

edit: on second look it appears that z hop is working. You need to disable reduce infill retraction in the process - other tab.

MrBlonde1989, 2023-12-23:
After unchecking this option, it seems to work fine.
[image]

[image]

Thank you, for you support

closed 2023-12-23 by MrBlonde1989