OrcaSlicer/OrcaSlicer #3224: Orca is taking extra 2h with unnecessary solid infill/shell thickness

HakunMatat4, 2023-12-21:
### OrcaSlicer Version

1.9.0-dev, 1.8.1, 1.7.0

### OS version

Linux Mint Cinnamon

### Printer

Voron Trident

### How to reproduce

1. Slice the STL attached
2. Check layer by layer to see the problem

### Actual results

**NOTE**: This has been happening for some time now, it just got worse.

~2h wasted with solid infill but shell thickness is disabled.
No explains is good enough to explain that, I usually print with 3 walls, you do not need any extra support, that is why I keep shell thickness disabled.

[image]

If I enable shell thickness, it does the opposite, it disables those non necessary infill and takes less time to print.

[image]

I tried to slice again and from 6h22 it changed to 6h10 so something weird is going on.
As I tried to replicate things to get the log, it went to 6h39

[image]

The screenshots above were shared within Discord, now that I am replicating everything to open this bug report, shell thickness is finally showing what it should have done in the the first place and 30 minutes longer.
There is some logic bug going on

From 6j22 to 6h39 to 7h, this cannot be right.

[image]

### Expected results

I expect no extra shell/infill to be added on its own.
It is taking an extra 2-3h to print, wasting filament and energy for no reason.

### Project file & Debug log uploads

[debug_log.txt](https://github.com/SoftFever/OrcaSlicer/files/13741866/debug_log.txt)
[Clockspring3D Torture Toaster v24.stl.zip](https://github.com/SoftFever/OrcaSlicer/files/13741870/Clockspring3D.Torture.Toaster.v24.stl.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

igiannakas, 2023-12-21:
I've replied to Discord for the above - but:
1. Two perimeters on sloped edges trigger the ensure horizontal thickness algorithm more frequently. This cannot be disabled from the GUI and it also has adverse effects on bridged surfaces so not recommended to create an option for it.
2. Enable detect narrow solid infill - it makes the print MUCH faster

HakunMatat4, 2023-12-22:
@igiannakas I disabled detect narrow solid infill because on 1.8.1, it was generating these weird unnecessary moves/concentric.
I just noticed that 1.9.0-dev is doing the same with that feature disabled.

[image]

I did a benchy test with detect narrow internal solid infill

2 Wall loops
[image]

2 Wall loops
[image]

Conclusion: Detect narrow internal solid infill somewhat decrease the time. Not much of a difference.
Now, I have been printing 3 walls lately, everything changes.

-----

3 Wall loops

[image]

3 Wall loops
[image]

Conclusion: With 3 walls, it prints 30s faster, still not a huge difference

-----

Back to the model I used to open this ticket.

3 Wall loops
[image]

3 Wall loops
[image]

Conclusion: Not much of a difference, 20 minutes

----

Last but not least...
This big project on the backlog...

3 Wall loops: **ON/OFF looks exactly the same like if I had not changed anything**
[image]

2 Wall loops: **ON/OFF looks exactly the same like if I had not changed anything**
[image]

igiannakas, 2023-12-22:
@HakunMatat4 what operating system are you running? I can send you a version with it completely disabled for testing as I’m hesitant to push a PR for this yet

HakunMatat4, 2023-12-22:
> @HakunMatat4 what operating system are you running? I can send you a version with it completely disabled for testing as I’m hesitant to push a PR for this yet

Hey, I'm on Linux. Thank you for looking into this.
I believe "features" like this should be added as option.
My problem with "ensure shell thickness" was that I always kept it disabled. That feature was removed and added back as active.

So I always believed the time and filament used was correct only to find out ensure thickness wall was added back enabled by default. I believe I mentioned that to noisefox, no such features should be enabled by force.

Smoothing you released, users enable it by choice. 
If you ever merge the PR understanding that UI changes must be a pain, I wonder if there's a way to only trigger it under certain circumstances like +3 walls for example must be used which provides enough strength, below that follows the current logic.

idk if that makes sense, you cover both worlds that way.

HakunMatat4, 2023-12-23:
I did a basic test and it looks awesome

I do have a question, this is your last test shared within the PR
[image]

This is mine and yet, yours is like 45 minutes faster. How low are flying that printer??? haha
[image]

Back to test, Detect narrow internal solid infill seems to be working better, also, that weird direction was driving me nuts, this PR makes it to improve and got rid of that printing in the air problem.

[image]

I will do some printing soon but so far the slicer shows no problem from what I can tell. I can print 2 walls again for prototyping prints without extra stuff being added, that is the main reason I was using 3 walls, to reduce some weird stuff.

igiannakas, 2023-12-23:
Just the standard BBL speeds with 0.2 layer height, adaptive cubic at 20% infill. The machine is indeed quick in the standard settings :)

Also suggest we move print tests to the PR itself

PR #3235 (2024-01-18): [Feature] Introduced experimental option to fully disable solid infill against walls when ensure vertical thickness is turned off
Introduced experimental option to fully disable solid infill printed against sloping walls even when ensure vertical thickness is turned off.

Fixes #3224
Fixes #3016
Fixes #1887

**Ensure vertical thickness ON:**
We can see solid infill against the walls even on pretty much flat surfaces. This is the most conservative option.
[image]

**Ensure vertical thickness OFF:**
Solid infill greatly reduced in mostly flat areas. 
[image]

However it is present on sharp slopi [... cut]
closed 2024-01-18 by SoftFever