OrcaSlicer/OrcaSlicer #3227: Slicer Failure "Error: One object has empty initial layer and can't be printed. Please Cut the bottom or enable supports"

weldmore, 2023-12-21:
### OrcaSlicer Version

1.8.1

### OS version

Windows 10

### Printer

Bambu X1c

### How to reproduce

Preview the attache file in vesion 1.8.1

### Actual results

"Error: One object has empty initial layer and can't be printed. Please Cut the bottom or enable supports"
Using Orca 1.8.1
Got this error when trying to slice the attached file for the Bambu X1C.  Nothing shows up in the sliced image and nothing will print. I did try cutting the model and adding supports but that did not help. But if I expand the model size by more than 30% in the X or Y direction it slices fine with no errors.
The attached file does slice in version 1.7.0. It reports an error "The model has too mant empty layers" but the sliced model prints fine.
[Wall Screen original.zip](https://github.com/SoftFever/OrcaSlicer/files/13743582/Wall.Screen.original.zip)

### Expected results

Should produce a sliced model

### Project file & Debug log uploads

[BS Wall Screen original.zip](https://github.com/SoftFever/OrcaSlicer/files/13743698/BS.Wall.Screen.original.zip)
[debug_network_Thu_Dec_21_10_53_43.zip](https://github.com/SoftFever/OrcaSlicer/files/13743713/debug_network_Thu_Dec_21_10_53_43.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

edeforas, 2023-12-30:
Same issue with the A1Mini, bugs also with a simple cylinder shell (enclosed)
[cylinder_shell.zip](https://github.com/SoftFever/OrcaSlicer/files/13798040/cylinder_shell.zip)

edeforas, 2023-12-31:
Same issue with the A1Mini, bugs also with a simple cylinder shell (enclosed)
[Cylinder_shell.zip](https://github.com/SoftFever/OrcaSlicer/files/13801012/Cylinder_shell.zip)

edeforas, 2024-01-06:
Same issue with Orca 1.9.0 ( thanks for the release!) , even with support enabled

xi2023, 2024-02-13:
I thought it was my model but no.. 

support and raft = error
only support = works
only raft = works
no support and no raft = works

looks like there is something wrong when you use tree support hybrid, slim or strong (even if your model doesnt needs supports) and a raft (that you probably would turn off if you could see the sliced preview). normal/default support settings did slice just fine.

Z18Aplha, 2024-02-27:
I had the same problem and it happened to me because I accidentally only entered "100" for "first layer". It is essential that "100%" is entered here, otherwise OrcaSlicer thinks it is 100mm.

marcinbauer85, 2024-05-14:
I have the same issue, but it was when I wanted to slice the model with 0.6 nozzle. When I changed to 0.4 it worked. Turns out the part that was causing the problem was too small for 0.6.

gunrunnerjohn, 2024-06-27:
This just happened to me.  I restarted the slicer and all was well and it sliced successfully.

closed 2024-10-03 by github-actions[bot]

inspectionsbybob, 2025-01-15:
I am having the same issue.  One Object works, multiple objects returns the "empty layer" error

gunrunnerjohn, 2025-01-15:
Orca is up to 2.2.0, I've had no significant issues with this release.

inspectionsbybob, 2025-01-15:
I sent a simple object on 2.2.0 that failed.   I am sure there is a conflict with some other setting, but I don't know what it is...  I don't really want to go through my entire profile, turning things off and trying to slice...  

Is there a LOG somewhere of what the slicer is seeing when it throws the error?

anoblet, 2025-02-01:
I just had this error today rotating a simple `cube [45, 45, 0]` and adding a brim.

kolik54, 2025-04-27:
Hi. I solved this problem by setting the minimum element size to 50% for Arachne. Before that it was 75% and the 0.35mm thick ring was not cut even with support and a raft. You can also choose the classic engine and everything will be cut without problems.