OrcaSlicer/OrcaSlicer #3238: <ctrl>-m dialog for setting mouse sensitivity is undiscoverable

cbaugher, 2023-12-22:
It would be nice to have a way to access the mouse settings without having to remember the keyboard shortcut.

foreachthing, 2023-12-23:
Why? I only need to adjust these settings _once_.
Just hit the "?" on your keyboard to bring up this dialog (don't have to remember anything 😉):
[image]

cbaugher, 2023-12-27:
Ok, that's fair.  A list of the hotkeys is available from the menu.  But if you didn't know that these settings even existed, how would you know to look for them in the hotkeys list?

Hotkeys are nice for people that like them.  But some of us don't like them.  I can do almost everything I need to in the software with just the trackball and space mouse.  Typing on the keyboard means I have to move my hands off of one of these devices.  That's annoying.

C|

closed 2024-04-03 by github-actions[bot]