OrcaSlicer/OrcaSlicer #3240: Add the ability to hide system printer preset when a User preset has been created

sprior, 2023-12-22:
Is your feature request related to a problem? Please describe.
I'd like to use BS for my Voron (in addition to my X1C), but it requires custom start gcode so I'd have to create a User preset based on the System Voron preset.

The problem is then I have two presets visible: the System preset I can NEVER use directly and the User preset. If someday I accidentally pick the System preset "bad" things will happen. I understand that BS requires the System preset to be there, but I'd like the ability to hide the System preset from the printer selection menu when I've created a User preset derived from it.

Super Slicer for example has a config preference named "Suppress - default - presets" - I believe the equivalent is exactly what I want.

I also opened this as a feature request on the Bambu Studio github (https://github.com/bambulab/BambuStudio/issues/3202) but it looks like the printer profile system for Orca has diverged from BS (giving it additional abilities) so I'm not sure which of you could get to  this first and whether if BS did it whether it would come to Orca automatically or not.

sprior, 2024-03-22:
This doesn't resonate with anyone?

skylatian, 2024-04-27:
Would like to bump this - I have a couple printers myself, and also manage a variety of other printers at a maker space. Having the extra profiles clogging up the UI is a pain, and I agree! If a custom preset has been created, the built-in one should be hidden.

sprior, 2024-07-27:
I'm still interested in this.

skylatian, 2024-07-27:
I have a fork that does this - barely.

I made it a toggle in the preferences pane. Re-showing the presets currently requires restarting Orcaslicer, but it doesn't *seem* to cause any other instabilities that I've found. Keep in mind I'm not a C++ dev (or a dev at all lol)

If I haven't updated it to the newest version of Orcaslicer in a week, ping me again and I'll make sure to take care of it!

As it stands, I believe [my last nightly](https://github.com/skylatian/OrcaSlicer-hide_presets/releases/tag/nighty-builds) works to at least hide them. I'm also not sure if that nighly is actually the 2.1.0 alpha or if it's an older version (~2.0.1 or so).

AlatusAles, 2024-11-01:
Up!

balthisar, 2025-01-24:
Still not stale.

DangerManiac, 2025-03-22:
If this is in PrusaSlice and has been for awhile, WTH is it not in OcraSlice? Sorry but a little irritated as I am trying to add a bunch of filaments and getting a little sick n tired of having to scroll through a bunch of presets to get to my profiles:

PrusaSlice:
[image]

Here I am trying to set the Machine Dependency, I have ONE Neptune 4, so why do I have to scroll through all of these:
[image]

Here I am trying to set Profile specific Dependency and I have to scroll through all these and MUCH more to get to my profiles:
[image]

hermitdafunk, 2025-04-21:
Bump. Very good feature to have, especially for those with a long list of presets.

Please and thank you.
Kind regards

Czarnyszarf, 2025-04-23:
up

sprior, 2025-07-23:
I'm still interested

Felix14-v2, 2026-10-06:
Implemented: you can use the `Detach from parent` option when saving a custom profile, and after that you can remove the parent without removing your custom profile. The only problem is that currently you have to save+detach it twice to apply the changes due to a bug, but at least there's an easy way to perform that now.

[image]

closed 2026-10-07 by ianalexis